eclipse-mraa/mraa #946: mraa/i2c_mpu6050.c acceleration only outputs integer

CODO77, 2019-01-05:
With the example you provide to read the accelerometer and gyroscope values from the MPU6050, it only outputs integer values of acceleration in terms of g. so if rounds the acceleration to the nearest whole number which can only be (-2,-1,0,1,2). I cannot find a way to display it as a decimal number, every time i try to convert it I get an extremely large integer number.

alext-mkrs, 2019-01-06:
I see - well, in this specific case the sensor data is not actually the point, the example is only used to demonstrate how to communicate using mraa via I2C, the specific thing we're talking _to_ is almost irrelevant.

However if you have a patch that would enhance the example - please feel free to send it as a PR, I'd be glad to review that (see CONTRIBUTING.md for details if you're up to that).

CODO77, 2019-01-06:
Thank you. I will see what I can do

closed 2019-01-12 by alext-mkrs