eclipse-mraa/mraa #947: Python SPI wrapper (spiFromDesc) not working on rpi

donnm, 2019-01-22:
The method spiFromDesc in the mraa Python module does not work (at least for Raspberry Pi).

```
$ ls /dev/spidev0.*
/dev/spidev0.0  /dev/spidev0.1
```

```
$ ipython
Python 2.7.13 (default, Sep 26 2018, 18:42:22) 
Type "copyright", "credits" or "license" for more information.

IPython 5.1.0 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]: import mraa as m

In [2]: m.spiFromDesc("spi-raw-0-0")
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-2-14e5700dc8ec> in <module>()
----> 1 m.spiFromDesc("spi-raw-0-0")

/usr/local/lib/python2.7/dist-packages/mraa.pyc in spiFromDesc(desc)
    655 
    656     """
--> 657     return _mraa.spiFromDesc(desc)
    658 
    659 def i2cFromDesc(desc):

ValueError: Invalid SPI context
```

From /var/log/syslog:
```
Jan 22 11:48:21 x libmraa[8723]: libmraa version v2.0.0-5-g7a1db24 initialised by user 'x' with EUID 1000
Jan 22 11:48:21 x libmraa[8723]: gpio: platform doesn't support chardev, falling back to sysfs
Jan 22 11:48:21 x libmraa[8723]: libmraa initialised for platform 'Raspberry Pi Model B+ Rev 1' of type 5
Jan 22 11:48:29 x libmraa[8723]: mraa_init_io: Invalid IO type given.
```

Following the call through to mraa_init_io it appears that the parsing fails in the helper functions, resulting in the error "Invalid SPI context".

I changed initIo() in api/mraa/common.hpp to:

```
template <class T>
inline T*
initIo(std::string desc)
{
//    return new T(mraa_init_io(desc.c_str()));
    return new T(mraa_spi_init(0));
}
```

bypassing the IO parser and hardcoding the SPI bus to 0, and it works fine.

alext-mkrs, 2019-02-03:
Okay, then looking at the code more closely, looks like the description parsing logic somehow doesn't recognize your description string - and specifically the very first part of it, `spi`.

Could you please try `s-raw-0-0` instead? Looks like we may have a bug in our docs/examples, as also recently reported in #948. I think ae391fe9fe79964b1cfa0d75918954a81a03fbf8 has changed those keys without updating all relevant pieces.

donnm, 2019-02-06:
I had previously tried many combinations of these strings.

```
# ls -l /dev/spidev0.*
crw-rw---- 1 root spi 153, 0 Jan 17 12:36 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 Jan 17 12:36 /dev/spidev0.1
# ipython
Python 2.7.13 (default, Sep 26 2018, 18:42:22) 
Type "copyright", "credits" or "license" for more information.

IPython 5.1.0 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]: import mraa as m

In [2]: m.spiFromDesc("s-raw-0-0")
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-2-45d2d4cc8f75> in <module>()
----> 1 m.spiFromDesc("s-raw-0-0")

/usr/local/lib/python2.7/dist-packages/mraa.pyc in spiFromDesc(desc)
    655 
    656     """
--> 657     return _mraa.spiFromDesc(desc)
    658 
    659 def i2cFromDesc(desc):

ValueError: Invalid SPI context

In [3]: m.spiFromDesc("0")
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-3-c1c4e11a974a> in <module>()
----> 1 m.spiFromDesc("0")

/usr/local/lib/python2.7/dist-packages/mraa.pyc in spiFromDesc(desc)
    655 
    656     """
--> 657     return _mraa.spiFromDesc(desc)
    658 
    659 def i2cFromDesc(desc):

ValueError: Invalid SPI context

```