eclipse-mraa/mraa #950: Query: What is the rate at which AIO read and GPIO write can be called

BinduRao2018, 2019-02-04:
i am using mraa on Raspberry PI with Grove Kit. 
I see occasional failures when I use mraa_aio_read (read_status = -1) and mraa_gpio_write (status = MRAA_ERROR_UNSPECIFIED).  I have several writes to GPIO and reads from AIO (various pins). All the calls are serialized and I invoke these apis continuously. I would like to know if there is any limitation w.r.t successive reads/writes using mraa. 
Can retry mechanism in these cases help or should i try writing to/reading from after a delay?

Mani-Sadhasivam, 2019-02-06:
Hi @BinduRao2018,

IMO this behavior is the limitation of SYSFS interface for both GPIO and AIO. You can test this by writing a simple shell script which just reads from AIO (/sys/bus/iio/devices/iio:device0/in_voltage0_raw) and writes to GPIO through SYSFS. I would suggest you to provide a delay after each read/write.

BinduRao2018, 2019-02-07:
Thanks for the info @Mani-Sadhasivam. I will look into this.

closed 2026-02-15 by alext-mkrs