eclipse-mraa/mraa #954: issue with uart devpath in tag 2.0.0

vsivanagulu, 2019-02-22:
I ma trying to add uart to my custom am335x based board . I have issues with devpath in uart initialization 

`mraa_result_t
mraa_cosmic_uart_init_pre(int index)
{
    char devpath[MAX_SIZE];
    snprintf(devpath,MAX_SIZE,"/dev/ttyO%u",index);
    if (!mraa_file_exist(devpath)) {
        syslog(LOG_ERR, "uart: Device not initialized");
    }else {
        plat->uart_dev[index].device_path = devpath;
        return MRAA_SUCCESS;
     }
    return MRAA_ERROR_INVALID_PARAMETER;
}
`
In above I have printed devpath values 

> else {
>        printf("path=%s",devpath);
>         plat->uart_dev[index].device_path = devpath;
>         printf("devpathaftre=%s",plat->uart_dev[index].device_path);
>         return MRAA_SUCCESS;
>      }
> output:
> path=/dev/ttyO1
> Failed to initialize UART
> devpathafter=/dev/ttyO1
> 

if I add like below then UART is working  fine
**'plat->uart_dev[index].device_path = "/dev/ttyO1";'**

Propanu, 2019-02-22:
Maybe check formatting/string termination on your written `devpath`? Just like you pointed out, this works with a constant literal so I doubt it's a mraa issue.

alext-mkrs, 2019-02-23:
@vsivanagulu, please post syslog contents around the time of you program run. The error message you post is not from the code you posted, but from UART example code, right? Posting full code you've changed may be helpful as well, but we can start with syslog.

alext-mkrs, 2019-05-05:
@vsivanagulu, is this still actual?

vsivanagulu, 2019-05-22:
@alext-mkrs ,i have stoped this currently for some other important work. i will do it soon.