eclipse-mraa/mraa #956: UP-Board / Ubilinux  : Onboard LEDs not working in mraa 2.0

lurabo, 2019-03-05:
I used to control UPBoard onboard LEDs with mraa 1.8 version many times, no issue.  I used c prog. language to control them.

Since I upgraded to mraa 2.0 version, LEDS stopped to work. 

I initialize them with this code :
led_context_green = mraa_led_init("green");
.... but  mraa_led_init("green") function returns 0 

_Here is mraa syslog message :_ 
mar 03 18:31:59 ubilinux4 libmraa[2576]: **libmraa version v2.0.0** initialised by user 'user123' with EUID 1000
mar 03 18:31:59 ubilinux4 libmraa[2576]: Adding i2c bus found on i2c-1 on adapter .
mar 03 18:31:59 ubilinux4 libmraa[2576]: Adding i2c bus found on i2c-0 on adapter .
mar 03 18:31:59 ubilinux4 libmraa[2576]: up: kernel pinctrl driver available
mar 03 18:31:59 ubilinux4 libmraa[2576]: gpio: platform doesn't support chardev, falling back to sysfs
mar 03 18:31:59 ubilinux4 libmraa[2576]: **libmraa initialised for platform 'UP' of type 12**
mar 03 18:32:11 ubilinux4 libmraa[2576]: **led: init: no led device defined in platform**

So, I suposse there is missing or changed UPboard LEDS definition in mraa 2.0 version ?

Propanu, 2019-03-05:
Hi @lurabo. The led APIs were updated in MRAA 1.9 as pointed out in the [changelog](../blob/master/docs/changelog.md). You will need to use the `mraa_led_init_raw(const char* led_name)` function instead in your code.

Here's a more interesting example that uses all the built-in LEDs on the UP2: https://github.com/intel-iot-devkit/iot-devkit-samples/blob/iss2019-update3/up2-leds/c/up2-leds.c

lurabo, 2019-03-06:
> You will need to use the `mraa_led_init_raw(const char* led_name)` function instead in your code.

Thank You for answer. It works now.

closed 2019-03-06 by lurabo