eclipse-mraa/mraa #957: mraa in Docker reports that user doesn't have access (when they do)

jveitchmichaelis, 2019-03-07:
I'm running a Raspberry Pi 3B+ with Raspbian Stretch. This works fine (outside Docker):

```
Python 2.7.13 (default, Sep 26 2018, 18:42:22) 
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa
>>> mraa.Led("led0")
<mraa.Led; proxy of <Swig Object of type 'mraa::Led *' at 0x76ad5470> >
>>> 
```
I've got a Docker container in which I've also installed mraa as per https://github.com/intel-iot-devkit/mraa/pull/928. I'm running the container with:

`docker run -it -P --privileged -v /sys:/sys bash`

When I try to run mraa (as root), I get the following:

```
Python 2.7.13 (default, Sep 26 2018, 18:42:22) 
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa
libmraa[167]: libmraa version v2.0.0-10-gb425c99 initialised by user 'root' with EUID 0
libmraa[167]: gpio: platform doesn't support chardev, falling back to sysfs
libmraa[167]: libmraa initialised for platform 'Raspberry Pi 3 Model B+' of type 5
>>> mraa.Led("led0")
libmraa[167]: led: init: current user don't have access rights for using LED led0
```

This stumped me for a bit, until I noticed that the LEDs were indeed controllable. So is mraa incorrectly determining permissions?

tingleby, 2019-03-28:
@Mani-Sadhasivam Any clue whats going on there :/
 https://github.com/intel-iot-devkit/mraa/blob/master/src/led/led.c#L163

jveitchmichaelis, 2019-03-28:
If I have some time I can compile in debug and see which condition it's failing. Within the container I can still write manually to the led device (via /sys) so I'm not sure why `stat` would fail.

Propanu, 2019-03-28:
I always get this on the UP2 too when running as non-root even if the user has write permission to the LED classes.