eclipse-mraa/mraa #961: Cannot import mraa in python3 (swig4 cmake3.13 RaspberryPI)

g-vidal, 2019-04-11:
I have compiled mraa on raspberryPi3 buster with swig4 and cmake version 3.13.4
Compilation is OK and installation goes well but even if the modules exist they cannot be imported :

```shell
raspife3-€ :  python3
Python 3.7.3rc1 (default, Mar 13 2019, 11:01:15) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa
>>> led=mraa.GPio(12)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'mraa' has no attribute 'GPio'
```
If I change `CMP0078` to  `OLD` the issue is solved.
I have proposed a pull request  #960 which might be rough but works.
```shell
python3
Python 3.7.3rc1 (default, Mar 13 2019, 11:01:15) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa
>>> led = mraa.Gpio(12)
>>> 
```

tingleby, 2019-04-11:
In your first output, mraa.Gpio has been typed with the wrong case on the 'p', could you please try without #960 but with the correct "casing"?
mraa.Gpio not mraa.G**P**io 

I am just curious if the CMP0078 has effected it :)

g-vidal, 2019-04-11:
Hum hum hum You are right!
It seemsto work except the error fpor the cmake. BUT I still have the error with upm.... any idea?

```shell
-- PYTHON2 attempting to build!
CMake Warning (dev) at /usr/share/cmake-3.13/Modules/UseSWIG.cmake:564 (message):
  Policy CMP0078 is not set.  Run "cmake --help-policy CMP0078" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  src/python/python2/CMakeLists.txt:9 (swig_add_library)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- PYTHON3 attempting to build!
CMake Warning (dev) at /usr/share/cmake-3.13/Modules/UseSWIG.cmake:564 (message):
  Policy CMP0078 is not set.  Run "cmake --help-policy CMP0078" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  src/python/python3/CMakeLists.txt:9 (swig_add_library)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- INFO - NODE_ROOT_DIR is /usr/local/include/node
-- Found Nodejs: /usr/local/bin/node (found version "11.13.0") 
-- Found v8: /usr/local/include/node/v8.h (found version "7.0.276.38-node.18")
CMake Warning (dev) at /usr/share/cmake-3.13/Modules/UseSWIG.cmake:564 (message):
  Policy CMP0078 is not set.  Run "cmake --help-policy CMP0078" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  src/javascript/CMakeLists.txt:27 (swig_add_library)
This warning is for project developers.  Use -Wno-dev to suppress it.
```