eclipse-mraa/mraa #963: UP Squared: getPin returns 0

gbr1, 2019-04-24:
Hi,
I'm using mraa v2.0.0 on Up Squared using C++ for a [ROS integration](https://github.com/gbr1/upboard_ros/tree/devel).
Everything is working but in my code I need to get pin number from a GPIO.
I can write on pin but result of **getPin()** is always 0 also for **getPin(true)**.
```
mraa::Gpio gpio_22(22);
status = gpio_22.dir(mraa::DIR_OUT);
if (status != mraa::SUCCESS) {
   printError(status);
   return EXIT_FAILURE;
}
```
Faulty line:<br>
`std::cout<<gpio_22.getPin()<<std::endl;`<br>
Thank you in advance!

Propanu, 2019-04-24:
Hi @gbr1. This might be caused by using the chardev devices instead of sysfs classes on the UP2. Pin addressing in this case works with a chip + line tuple.
Perhaps the common `getPinName(int pin)` function can help instead? (see http://iotdk.intel.com/docs/master/mraa/namespacemraa.html)
If not, you'll have to compile MRAA from source and disable chardev for the platform.

gbr1, 2019-04-24:
Hi @Propanu! Just checked. <br>
`std::cout<<mraa::getPinName(22);`<br>
give me **GPIO25** and I assume it should be correct because [here](https://wiki.up-community.org/Pinout_UP2) it is called _BCM25_.
I solved inheriting Gpio class to a new Gpio class that override getPin method because I'm trying to make easy to install my ROS wrapper (just a git clone and catkin_make).
I don't know if this issue could be useful for future updates.
Thank you!

alext-mkrs, 2019-05-05:
This looks like resolved to me  - closing. Feel free to post if you need further help.

closed 2019-05-05 by alext-mkrs