eclipse-mraa/mraa #969: Build failure with Node.js 12.x

sgallagher, 2019-06-19:
When building against Node.js 12.4.0 on Fedora Rawhide, we get the following output:

```
[ 33%] Built target mraa
make -f src/python/python3/CMakeFiles/_python3-mraa.dir/build.make src/python/python3/CMakeFiles/_python3-mraa.dir/depend
make -f src/javascript/CMakeFiles/mraajs.dir/build.make src/javascript/CMakeFiles/mraajs.dir/depend
make[2]: Entering directory '/builddir/build/BUILD/mraa-2.0.0'
cd /builddir/build/BUILD/mraa-2.0.0 && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /builddir/build/BUILD/mraa-2.0.0 /builddir/build/BUILD/mraa-2.0.0/src/javascript /builddir/build/BUILD/mraa-2.0.0 /builddir/build/BUILD/mraa-2.0.0/src/javascript /builddir/build/BUILD/mraa-2.0.0/src/javascript/CMakeFiles/mraajs.dir/DependInfo.cmake --color=
make[2]: Entering directory '/builddir/build/BUILD/mraa-2.0.0'
cd /builddir/build/BUILD/mraa-2.0.0 && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /builddir/build/BUILD/mraa-2.0.0 /builddir/build/BUILD/mraa-2.0In file included from /builddir/build/BUILD/mraa-2.0.0/src/python/python3/CMakeFiles/_python3-mraa.dir/mraa3PYTHON_wrap.cxx:3069:
/builddir/build/BUILD/mraa-2.0.0/api/mraa/gpio.hpp: In member function 'mraa::Result mraa::Gpio::useMmap(bool)':
/builddir/build/BUILD/mraa-2.0.0/api/mraa/gpio.hpp:317:77: warning: 'mraa_result_t mraa_gpio_use_mmaped(mraa_gpio_context, mraa_boolean_t)' is deprecated [-Wdeprecated-declarations]
  317 |         return (Result) mraa_gpio_use_mmaped(m_gpio, (mraa_boolean_t) enable);
      |                                                                             ^
In file included from /builddir/build/BUILD/mraa-2.0.0/api/mraa/gpio.hpp:27,
                 from /builddir/build/BUILD/mraa-2.0.0/src/python/python3/CMakeFiles/_python3-mraa.dir/mraa3PYTHON_wrap.cxx:3069:
/builddir/build/BUILD/mraa-2.0.0/api/mraa/gpio.h:279:26: note: declared here
  279 | DEPRECATED mraa_result_t mraa_gpio_use_mmaped(mraa_gpio_context dev, mraa_boolean_t mmap);
      |                          ^~~~~~~~~~~~~~~~~~~~
In file included from /builddir/build/BUILD/mraa-2.0.0/src/python/python3/CMakeFiles/_python3-mraa.dir/mraa3PYTHON_wrap.cxx:3069:
/builddir/build/BUILD/mraa-2.0.0/api/mraa/gpio.hpp:317:77: warning: 'mraa_result_t mraa_gpio_use_mmaped(mraa_gpio_context, mraa_boolean_t)' is deprecated [-Wdeprecated-declarations]
  317 |         return (Result) mraa_gpio_use_mmaped(m_gpio, (mraa_boolean_t) enable);
      |                                                                             ^
In file included from /builddir/build/BUILD/mraa-2.0.0/api/mraa/gpio.hpp:27,
                 from /builddir/build/BUILD/mraa-2.0.0/src/python/python3/CMakeFiles/_python3-mraa.dir/mraa3PYTHON_wrap.cxx:3069:
/builddir/build/BUILD/mraa-2.0.0/api/mraa/gpio.h:279:26: note: declared here
  279 | DEPRECATED mraa_result_t mraa_gpio_use_mmaped(mraa_gpio_context dev, mraa_boolean_t mmap);
      |                          ^~~~~~~~~~~~~~~~~~~~
/builddir/build/BUILD/mraa-2.0.0/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx: In function 'v8::Handle<v8::Value> SWIG_V8_NewPointerObj(void*, swig_type_info*, int)':
/builddir/build/BUILD/mraa-2.0.0/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx:1188:79: error: no matching function for call to 'v8::ObjectTemplate::NewInstance()'
 1188 |   v8::Local<v8::Object> result = class_templ->InstanceTemplate()->NewInstance();
      |                                                                               ^
In file included from /usr/include/node/node.h:63,
                 from /builddir/build/BUILD/mraa-2.0.0/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx:171:
/usr/include/node/v8.h:6244:44: note: candidate: 'v8::MaybeLocal<v8::Object> v8::ObjectTemplate::NewInstance(v8::Local<v8::Context>)'
 6244 |   V8_WARN_UNUSED_RESULT MaybeLocal<Object> NewInstance(Local<Context> context);
      |                                            ^~~~~~~~~~~
/usr/include/node/v8.h:6244:44: note:   candidate expects 1 argument, 0 provided
/builddir/build/BUILD/mraa-2.0.0/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx: In function 'void SWIGV8_AddStaticFunction(v8::Handle<v8::Object>, const char*, void (* const&)(const v8::FunctionCallbackInfo<v8::Value>&))':
/builddir/build/BUILD/mraa-2.0.0/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx:1513:83: error: no matching function for call to 'v8::FunctionTemplate::GetFunction()'
 1513 |   obj->Set(SWIGV8_SYMBOL_NEW(symbol), SWIGV8_FUNCTEMPLATE_NEW(_func)->GetFunction());
      |                                                                                   ^
In file included from /usr/include/node/node.h:63,
                 from /builddir/build/BUILD/mraa-2.0.0/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx:171:
/usr/include/node/v8.h:5947:46: note: candidate: 'v8::MaybeLocal<v8::Function> v8::FunctionTemplate::GetFunction(v8::Local<v8::Context>)'
 5947 |   V8_WARN_UNUSED_RESULT MaybeLocal<Function> GetFunction(
      |                                              ^~~~~~~~~~~
/usr/include/node/v8.h:5947:46: note:   candidate expects 1 argument, 0 provided
/builddir/build/BUILD/mraa-2.0.0/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx: In function 'void SWIGV8_AddStaticVariable(v8::Handle<v8::Object>, const char*, SwigV8AccessorGetterCallback, SwigV8AccessorSetterCallback)':
/builddir/build/BUILD/mraa-2.0.0/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx:1524:87: warning: ignoring return value of 'v8::Maybe<bool> v8::Object::SetAccessor(v8::Local<v8::Context>, v8::Local<v8::Name>, v8::AccessorNameGetterCallback, v8::AccessorNameSetterCallback, v8::MaybeLocal<v8::Value>, v8::AccessControl, v8::PropertyAttribute, v8::SideEffectType, v8::SideEffectType)', declared with attribute warn_unused_result [-Wunused-result]
 1524 |   obj->SetAccessor(SWIGV8_CURRENT_CONTEXT(), SWIGV8_SYMBOL_NEW(symbol), getter, setter);
      |                                                                                       ^
In file included from /usr/include/node/node.h:63,
                 from /builddir/build/BUILD/mraa-2.0.0/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx:171:
/usr/include/node/v8.h:3463:37: note: declared here
 3463 |   V8_WARN_UNUSED_RESULT Maybe<bool> SetAccessor(
      |                                     ^~~~~~~~~~~
In file included from /builddir/build/BUILD/mraa-2.0.0/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx:1776:
/builddir/build/BUILD/mraa-2.0.0/api/mraa/gpio.hpp: In static member function 'static void mraa::Gpio::v8isr(uv_work_t*, int)':
/builddir/build/BUILD/mraa-2.0.0/api/mraa/gpio.hpp:176:63: error: no matching function for call to 'v8::Function::Call(v8::Local<v8::Object>, int&, v8::Local<v8::Value> [1])'
  176 |         f->Call(SWIGV8_CURRENT_CONTEXT()->Global(), argc, argv);
      |                                                               ^
In file included from /usr/include/node/node.h:63,
                 from /builddir/build/BUILD/mraa-2.0.0/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx:171:
/usr/include/node/v8.h:4148:43: note: candidate: 'v8::MaybeLocal<v8::Value> v8::Function::Call(v8::Local<v8::Context>, v8::Local<v8::Value>, int, v8::Local<v8::Value>*)'
 4148 |   V8_WARN_UNUSED_RESULT MaybeLocal<Value> Call(Local<Context> context,
      |                                           ^~~~
/usr/include/node/v8.h:4148:43: note:   candidate expects 4 arguments, 3 provided
In file included from /builddir/build/BUILD/mraa-2.0.0/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx:1776:
/builddir/build/BUILD/mraa-2.0.0/api/mraa/gpio.hpp: In member function 'mraa::Result mraa::Gpio::useMmap(bool)':
[... 90 lines cut ...]
make[2]: *** [src/javascript/CMakeFiles/mraajs.dir/build.make:66: src/javascript/CMakeFiles/mraajs.dir/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:327: src/javascript/CMakeFiles/mraajs.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
.0/src/python/python3 /builddir/build/BUILD/mraa-2.0.0 /builddir/build/BUILD/mraa-2.0.0/src/python/python3 /builddir/build/BUILD/mraa-2.0.0/src/python/python3/CMakeFiles/_python3-mraa.dir/DependInfo.cmake --color=
Scanning dependencies of target mraajs
make[2]: Leaving directory '/builddir/build/BUILD/mraa-2.0.0'
make -f src/javascript/CMakeFiles/mraajs.dir/build.make src/javascript/CMakeFiles/mraajs.dir/build
[... cut ...] HON3=TRUE -o CMakeFiles/_python3-mraa.dir/CMakeFiles/_python3-mraa.dir/mraa3PYTHON_wrap.cxx.o -c /builddir/build/BUILD/mraa-2.0.0/src/python/python3/CMakeFiles/_python3-mraa.dir/mraa3PYTHON_wrap.cxx
make[2]: Leaving directory '/builddir/build/BUILD/mraa-2.0.0'
```

The noteworthy piece is:
```
candidate expects 1 argument, 0 provided
```

Node.js 12.x moved to running atop version 7.4 of v8, which is a backwards-incompatible major version upgrade from Node.js 10.x (which used version 6.8 of v8).

Propanu, 2019-06-29:
Hi @sgallagher, maybe follow [this](https://github.com/swig/swig/issues/1520) for now, as I don't think there is a swig patch ready yet.

g-vidal, 2019-12-05:
Hi @Propanu @sgallagher ,
It does not work either it is better but  the compilation fails a bit further  : 

`
[ 37%] Building CXX object src/javascript/CMakeFiles/mraajs.dir/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx.o
/home/ens-ife/Logiciels/IoT/mraa/build/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx: In function ‘v8::Handle<v8::Value> SWIG_V8_NewPointerObj(void*, swig_type_info*, int)’:
/home/ens-ife/Logiciels/IoT/mraa/build/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx:1188:79: error: no matching function for call to ‘v8::ObjectTemplate::NewInstance()’
 1188 |   v8::Local<v8::Object> result = class_templ->InstanceTemplate()->NewInstance();
      |                                                                               ^
`

I have tried some unsuccessfull tricks but I am not able to contribute on this  ;-( . As nodejs is rushing to newer and newer versions node.js 13;x is out we may need to call for support because if we stay stucked on nodejs 10.x all this fantastic software may loose part of its interest. I can help with checking/testing but unfortunately no more. Thanks for this work, hope a solution will come up....

nullr0ute, 2019-12-05:
> I have tried some unsuccessfull tricks but I am not able to contribute on this ;-( . As nodejs is rushing to newer and newer versions node.js 13;x is out we may need to call for support because if [...]

This is causing us a lot of issues in Fedora, and given there's been no movement upstream and frankly this project appears almost dead, I'm considering just retiring it all.

g-vidal, 2019-12-05:
The problem comes from the interaction between node-swig-C++ . I understand that it requires advanced skills but I do not know any equivalent resource to propose a learning platform for IoT or any type of sensor or any GPIO connected tool enabling programming in three languages. We use Jupyter  on raspberries and it is fun and efficiency. It would be very sad that this project vanishes.

tingleby, 2019-12-06:
The Swig/Node integration has always been quite fragile. I've been a bit busy with other stuff recent and haven't been able to give much time to mraa ontop of the ongoing move to the Eclipse Foundation.

g-vidal, 2019-12-09:
OK thanks for the job  anyway. If you tackle the problem let me know. 
My feeling is that the problem comes from a function call `(v8::ObjectTemplate::NewInstance())` that dos not follow the template picked in V8.  It is amatter of context but a adding the context make the compilation fall into a series of bugs. I do not if it could be solved only by changing mraa code or if an action on swig is necessary. I read some exchanges on swig issues concerning NewInstance() then I dropped off due to lack of skills in nodeJS.

g-vidal, 2020-01-04:
Hi @tingleby just a little reminder on that point nothing seems to move on swig side I tested the compilation on a raspi4 with swwig4.0.2 and NodeJS13.5.0 
The couple  4.0.1 / 10.16.3    is still working (at least for mraa upm compilation I have not yet tested the tools).
If needed I can provide  the output with the errors but I thing anybody canget it.
happy new year to all...

nullr0ute, 2020-02-25:
Trying the patch from [1] with mraa 2.1.0 it still fails with newer NodeJS :(

[1] https://github.com/swig/swig/pull/1702

lissyx, 2020-02-25:
> Trying the patch from [1] with mraa 2.1.0 it still fails with newer NodeJS :(
> 
> [1] [swig/swig#1702](https://github.com/swig/swig/pull/1702)

As documented on this PR / issue, my patch is incomplete because I've only tested on our usecase. Feel free to help getting the PR to work properly with Travis to help getting more coverage.

yegorich, 2020-06-08:
The main issue for me is the `f->Call()` invocation:

    f->Call(SWIGV8_CURRENT_CONTEXT()->Global(), argc, argv);

In the recent Node.js versions it requires 4 parameters instead of 3 (see [API](https://v8docs.nodesource.com/node-4.8/d5/d54/classv8_1_1_function.html#a9c3d0e4e13ddd7721fce238aa5b94a11))

The errors related to the Buffer can be workaround via just invoking the code parts for `SWIG_V8_VERSION > 0x040000`

Propanu, 2020-09-29:
@nxhack is this with the latest version of SWIG, or does it also require the unmerged patches I mentioned earlier on this issue?

Since the latest versions of node don't compile out of the box we decided to disable node.js builds by default for the next release. Easy to undo in a future release when all the needed bits are in place. Thanks.

nxhack, 2020-09-30:
@Propanu 

This is a case of applying the patch we are considering in SWIG.
https://github.com/swig/swig/pull/1746

I've applied these patches to the OpenWrt packages I maintain and so far there don't seem to be any reports of problems. (I don't know how many users are out there.)
https://github.com/openwrt/packages/blob/master/utils/swig/patches/000-support_v12.patch
https://github.com/openwrt/packages/blob/master/libs/libmraa/patches/020-support_v12.patch

(This OpenWrt package is intended for node v12 and above.)

g-vidal, 2020-10-26:
Hi @Propanu @nxhack ,Thanks for your work,  I tried your patch on current master `mraa`  with current node (node-v15.0.1-linux-armv7l) and current swig (Version: 4.1.0), compiled on raspberyPi debian buster  (Linux raspife3 4.19.127-v7 #1 SMP Sun Oct 4 15:17:59 CEST 2020 armv7l GNU/Linux) 

Compilation fails with 45 errors  (17 first occurences are listed below), I am not able to  build patches but I can test solutions if any proposal. It seems that _v8::MaybeLocal<v8::String>_ errors disappear using `ToLocalChecked()`. IT is strange that some **Handle<v8::Value>&** are still there.

```shell
mraajsJAVASCRIPT_wrap.cxx:903:69: error: no matching function for call to ‘v8::Exception::Error(v8::MaybeLocal<v8::String>)’
mraajsJAVASCRIPT_wrap.cxx: In member function ‘virtual void OverloadErrorHandler::error(int, const char*)’:
mraajsJAVASCRIPT_wrap.cxx:1137:17: error: ‘class v8::Persistent<v8::Object>’ has no member named ‘MarkIndependent’
mraajsJAVASCRIPT_wrap.cxx:1188:79: error: no matching function for call to ‘v8::ObjectTemplate::NewInstance()’
mraajsJAVASCRIPT_wrap.cxx:1285:128: error: no matching function for call to ‘v8::Private::ForApi(v8::Isolate*, v8::MaybeLocal<v8::String>)’
mraajsJAVASCRIPT_wrap.cxx:887:85: error: no matching function for call to ‘v8::Value::BooleanValue(v8::Local<v8::Context>)’
mraajsJAVASCRIPT_wrap.cxx:1412:17: error: ‘class v8::Persistent<v8::Object>’ has no member named ‘MarkIndependent’
mraajsJAVASCRIPT_wrap.cxx:1444:30: error: no matching function for call to ‘v8::Array::Set(uint32_t, v8::Handle<v8::Value>&)’
mraajsJAVASCRIPT_wrap.cxx:1476:56: error: no matching function for call to ‘v8::FunctionTemplate::SetClassName(v8::MaybeLocal<v8::String>)’
mraajsJAVASCRIPT_wrap.cxx:1482:96: error: no matching function for call to ‘v8::ObjectTemplate::Set(v8::MaybeLocal<v8::String>, v8::Local<v8::FunctionTemplate>)’
mraajsJAVASCRIPT_wrap.cxx:1505:69: error: no matching function for call to ‘v8::ObjectTemplate::SetAccessor(v8::MaybeLocal<v8::String>, void (*&)(v8::Local<v8::Name>, const v8::PropertyCallbackInfo<v8::Value>&), void (*&)(v8::Local<v8::Name>, v8::Local<v8::Value>, const v8::PropertyCallbackInfo<void>&))’
mraajsJAVASCRIPT_wrap.cxx:1513:83: error: no matching function for call to ‘v8::FunctionTemplate::GetFunction()’
mraajsJAVASCRIPT_wrap.cxx:1524:87: error: no matching function for call to ‘v8::Object::SetAccessor(v8::Local<v8::Context>, v8::MaybeLocal<v8::String>, void (*&)(v8::Local<v8::Name>, const v8::PropertyCallbackInfo<v8::Value>&), void (*&)(v8::Local<v8::Name>, v8::Local<v8::Value>, const v8::PropertyCallbackInfo<void>&))’
mraajsJAVASCRIPT_wrap.cxx:785:108: error: ‘kNormalString’ is not a member of ‘v8::String’
mraajsJAVASCRIPT_wrap.cxx:12395:30: error: ‘class v8::FunctionTemplate’ has no member named ‘SetHiddenPrototype’; did you mean ‘RemovePrototype’?
mraajsJAVASCRIPT_wrap.cxx:12396:91: error: no matching function for call to ‘v8::FunctionTemplate::GetFunction()’
mraajsJAVASCRIPT_wrap.cxx:12485:76: error: no matching function for call to ‘v8::Object::Set(v8::MaybeLocal<v8::String>, v8::Handle<v8::Object>&)’
```

nxhack, 2021-04-29:
Lack of "SWIG_V8_VERSION" consistency seems to be the root cause of this problem.

https://github.com/eclipse/mraa/issues/1040#issuecomment-828957352

nxhack, 2021-05-03:
While checking the handling of "SWIG_V8_VERSION" on the SWIG side, I made a patch for mraa and upm.

SWIG:
https://github.com/swig/swig/pull/2011#issuecomment-830914401

MRAA:
https://github.com/eclipse/mraa/pull/1064

UPM:
https://github.com/eclipse/upm/pull/703