eclipse-mraa/mraa #970: Compiler swap messes up includes

Erikvv, 2019-06-30:
I'm probably doing something wrong but I don't know how to proceed. Would like a pointer if anyone knows.

I can compile the library. But now I want to cross-compile to ARMv8.

When i change the build command from 

`cmake ..`

to 

`cmake .. -DCMAKE_C_COMPILER="aarch64-linux-gnu-gcc"`

The compiler command changes from 

```
/usr/bin/cc 
    -Dmraa_EXPORTS 
    -I/home/erik/projects/dragonboard-sensors/vendor/mraa/api 
    -I/home/erik/projects/dragonboard-sensors/vendor/mraa/api/mraa 
    -I/home/erik/projects/dragonboard-sensors/vendor/mraa/include 
    -I/usr/include/json-c  
    -Wall 
    -Werror=main 
    -Wformat 
    -Wmain 
    -Wuninitialized 
    -Winit-self 
    -Werror=implicit 
    -Werror=missing-parameter-type 
    -DONEWIRE=1 
    -DX86PLAT=1 
    -DJSONPLAT=1 
    -fPIC   
    -std=gnu99 
    -o CMakeFiles/mraa.dir/mraa.c.o   
    -c /home/erik/projects/dragonboard-sensors/vendor/mraa/src/mraa.c
```

to

```
/usr/bin/aarch64-linux-gnu-gcc 
    -Dmraa_EXPORTS 
    -I/usr/include
    -I/home/erik/projects/dragonboard-sensors/vendor/mraa/api 
    -I/home/erik/projects/dragonboard-sensors/vendor/mraa/api/mraa 
    -I/home/erik/projects/dragonboard-sensors/vendor/mraa/include 
    -I/usr/include/json-c  
    -Wall 
    -Werror=main 
     -Wformat 
    -Wmain 
    -Wuninitialized 
    -Winit-self 
    -Werror=implicit 
    -Werror=missing-parameter-type 
    -DONEWIRE=1 
    -DX86PLAT=1 
    -DJSONPLAT=1 
    -fPIC   
    -std=gnu99 
    -o CMakeFiles/mraa.dir/mraa.c.o   
    -c /home/erik/projects/dragonboard-sensors/vendor/mraa/src/mraa.c
```

The added `-I/usr/include` will cause the wrong headers to be included and compilation will fail.

Erikvv, 2019-06-30:
This is fixed if I remove line https://github.com/intel-iot-devkit/mraa/blob/0a12c5a0177f9fa8e7c4c564e70a65c4a0bb81af/CMakeLists.txt#L143

tingleby, 2019-07-01:
When cross compiling, you might need to specify a cmake toolchain file.