eclipse-mraa/mraa #971: RPI 4 support

Propanu, 2019-07-09:
Has anyone tried MRAA on the Raspberry Pi 4 yet? @Mani-Sadhasivam, @arfoll do you have any idea if we need to update the existing definition files? I don't have one of the new boards. Thanks!

Mani-Sadhasivam, 2019-07-10:
@Propanu I don't have any info about Rpi 4. Maybe I can ask @nullr0ute to help on this. AFAIK he is working on bringing Fedora on Rpi 4 and having MRAA support would be very beneficial.

bozkurthan, 2019-07-20:
I have raspberry pi 4, I can hand for you. Can you assign me for this issue ?

Mani-Sadhasivam, 2019-07-20:
@bozkurthan Awesome! Please go ahead and submit the PR. Thanks

Chuckduey, 2019-07-21:
I already had a version that I was working on.  I just put in a PR.  It just enables the basic detection so the 40 in version and the correct pin map is detected.  It does not have the advanced overlays.  It does detect the 3 different versions 1GB, 2GB and 4GB have different revision codes.  I mapped them all to the same name.  If you want to expand on this or have a better version, please be my guest.

bozkurthan, 2019-07-21:
I guess I'm not as expert as you, @Chuckduey . I examined your PR and I can't write better than you :) Please go ahead. I can test with 2 GB if you wish.

bozkurthan, 2019-07-30:
@Propanu are we ok about @Chuckduey PR?  #975

Propanu, 2019-07-30:
#975 looks good but unfortunately I can't merge it here anymore. @Chuckduey would it be possible to open this PR on the new official Eclipse MRAA repo?
https://github.com/eclipse/mraa
You should be able to simply update your git remote and reopen. Thanks!

BinduRao2018, 2020-10-28:
@Propanu, @Chuckduey - Are these changes merged? Whicn tagged version of mraa I should refer to? At present i am using libmraa on Raspberry Pi 3b+. I would like to know if I can start using Raspberry Pi 4

Propanu, 2020-10-29:
Hi @BinduRao2018, use our latest tag merged earlier today `v2.2.0` and it should include all RPI4 board detection patches we received so far. The base model was added since 2.1 so I guess we can close this as the board should be reasonably supported already. If you're running into issues let's open a new thread. Thanks!

closed 2020-10-29 by Propanu