eclipse-mraa/mraa #972: Error trying to compile on RockPi 4B with Ubuntu Mate Desktop

rolly-ng, 2019-07-11:
Hi Everyone,
I am trying to compile the mraa library on my RockPi 4 which I was told to support mraa library. Before that I tried to install with apt-get but I have no pins with mraa-gpio list.
Anyway, here are the output of errors, could anyone please take a look and advise?

rock@rockpi:~/Downloads/libmraa/build$ make
Scanning dependencies of target mraa
[ 1%] Building C object src/CMakeFiles/mraa.dir/mraa.c.o
[ 2%] Building C object src/CMakeFiles/mraa.dir/gpio/gpio.c.o
[ 3%] Building C object src/CMakeFiles/mraa.dir/gpio/gpio_chardev.c.o
[ 4%] Building C object src/CMakeFiles/mraa.dir/i2c/i2c.c.o
[ 5%] Building C object src/CMakeFiles/mraa.dir/pwm/pwm.c.o
[ 6%] Building C object src/CMakeFiles/mraa.dir/spi/spi.c.o
[ 8%] Building C object src/CMakeFiles/mraa.dir/aio/aio.c.o
[ 9%] Building C object src/CMakeFiles/mraa.dir/uart/uart.c.o
[ 10%] Building C object src/CMakeFiles/mraa.dir/led/led.c.o
/home/rock/Downloads/libmraa/src/led/led.c: In function ‘mraa_led_get_trigfd’:
/home/rock/Downloads/libmraa/src/led/led.c:46:33: warning: ‘%s’ directive output may be truncated writing 7 bytes into a region of size between 0 and 63 [-Wformat-truncation=]
snprintf(buf, MAX_SIZE, “%s/%s”, dev->led_path, “trigger”);
^~ ~~~~~~~~~
/home/rock/Downloads/libmraa/src/led/led.c:46:5: note: ‘snprintf’ output between 9 and 72 bytes into a destination of size 64
snprintf(buf, MAX_SIZE, “%s/%s”, dev->led_path, “trigger”);
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/rock/Downloads/libmraa/src/led/led.c: In function ‘mraa_led_get_brightfd’:
/home/rock/Downloads/libmraa/src/led/led.c:61:33: warning: ‘%s’ directive output may be truncated writing 10 bytes into a region of size between 0 and 63 [-Wformat-truncation=]
snprintf(buf, MAX_SIZE, “%s/%s”, dev->led_path, “brightness”);
^~ ~~~~~~~~~~~~
/home/rock/Downloads/libmraa/src/led/led.c:61:5: note: ‘snprintf’ output between 12 and 75 bytes into a destination of size 64
snprintf(buf, MAX_SIZE, “%s/%s”, dev->led_path, “brightness”);
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/rock/Downloads/libmraa/src/led/led.c: In function ‘mraa_led_get_maxbrightfd’:
/home/rock/Downloads/libmraa/src/led/led.c:76:33: warning: ‘%s’ directive output may be truncated writing 14 bytes into a region of size between 0 and 63 [-Wformat-truncation=]
snprintf(buf, MAX_SIZE, “%s/%s”, dev->led_path, “max_brightness”);
^~ ~~~~~~~~~~~~~~~~
/home/rock/Downloads/libmraa/src/led/led.c:76:5: note: ‘snprintf’ output between 16 and 79 bytes into a destination of size 64
snprintf(buf, MAX_SIZE, “%s/%s”, dev->led_path, “max_brightness”);
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 11%] Building C object src/CMakeFiles/mraa.dir/initio/initio.c.o
[ 12%] Building C object src/CMakeFiles/mraa.dir/grovepi/grovepi.c.o
[ 13%] Building C object src/CMakeFiles/mraa.dir/uart_ow/uart_ow.c.o
[ 14%] Building C object src/CMakeFiles/mraa.dir/iio/iio.c.o
/home/rock/Downloads/libmraa/src/iio/iio.c: In function ‘mraa_iio_get_channel_data’:
/home/rock/Downloads/libmraa/src/iio/iio.c:182:44: warning: format ‘%d’ expects a matching ‘int’ argument [-Wformat=]
syslog(LOG_ERR, “iio: Channel %d with channel bytes value <= 0”);
~^
/home/rock/Downloads/libmraa/src/iio/iio.c:40:26: warning: ‘%s’ directive output may be truncated writing up to 255 bytes into a region of size between 71 and 81 [-Wformat-truncation=]
#define IIO_SYSFS_DEVICE “/sys/bus/iio/devices/” IIO_DEVICE
^
/home/rock/Downloads/libmraa/src/iio/iio.c:113:37: note: in expansion of macro ‘IIO_SYSFS_DEVICE’
snprintf(buf, MAX_SIZE, IIO_SYSFS_DEVICE “%d/” IIO_SCAN_ELEM “/%s”, dev->num, ent->d_name);
^~~~~~~~~~~~~~~~
/home/rock/Downloads/libmraa/src/iio/iio.c:612:80: note: format string is defined here
snprintf(buf, MAX_SIZE, IIO_SYSFS_DEVICE “%d/” IIO_SCAN_ELEM “/%s”, dev->num, ent->d_name);
^~
/home/rock/Downloads/libmraa/src/iio/iio.c:113:13: note: ‘snprintf’ output between 48 and 313 bytes into a destination of size 128
snprintf(buf, MAX_SIZE, IIO_SYSFS_DEVICE “%d/” IIO_SCAN_ELEM “/%s”, dev->num, ent->d_name);
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/rock/Downloads/libmraa/src/iio/iio.c: In function ‘mraa_iio_get_event_data’:
/home/rock/Downloads/libmraa/src/iio/iio.c:40:26: warning: ‘%s’ directive output may be truncated writing up to 255 bytes into a region of size between 78 and 88 [-Wformat-truncation=]
#define IIO_SYSFS_DEVICE “/sys/bus/iio/devices/” IIO_DEVICE
^
/home/rock/Downloads/libmraa/src/iio/iio.c:416:41: note: in expansion of macro ‘IIO_SYSFS_DEVICE’
snprintf(buf, MAX_SIZE, IIO_SYSFS_DEVICE “%d/” IIO_EVENTS “/%s”, dev->num, ent->d_name);
^~~~~~~~~~~~~~~~
/home/rock/Downloads/libmraa/src/iio/iio.c:38:31: note: format string is defined here
#define IIO_SCAN_ELEM “scan_elements”
^~
/home/rock/Downloads/libmraa/src/iio/iio.c:416:17: note: ‘snprintf’ output between 41 and 306 bytes into a destination of size 128
snprintf(buf, MAX_SIZE, IIO_SYSFS_DEVICE “%d/” IIO_EVENTS “/%s”, dev->num, ent->d_name);
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/rock/Downloads/libmraa/src/iio/iio.c: In function ‘mraa_iio_update_channels’:
[... 273 lines cut ...]
^
/home/rock/Downloads/libmraa/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx: In function ‘void _wrap_delete_Led(int)’:
/home/rock/Downloads/libmraa/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:10497:40: error: ‘data’ was not declared in this scope
v8::Localv8::Object object = data.GetValue();
^~~~
/home/rock/Downloads/libmraa/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:10497:40: note: suggested alternative: ‘atan’
v8::Localv8::Object object = data.GetValue();
^~~~
atan
/home/rock/Downloads/libmraa/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx: In function ‘void SWIG_V8_SetModule(void*, swig_module_info*)’:
/home/rock/Downloads/libmraa/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:10875:15: error: ‘class v8::Object’ has no member named ‘SetHiddenValue’; did you mean ‘IntegerValue’?
global_obj->SetHiddenValue(SWIGV8_STRING_NEW(“swig_module_info_data”), mod);
^~~~~~~~~~~~~~
IntegerValue
/home/rock/Downloads/libmraa/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx: In function ‘swig_module_info* SWIG_V8_GetModule(void*)’:
/home/rock/Downloads/libmraa/build/src/javascript/mraajsJAVASCRIPT_wrap.cxx:10881:49: error: ‘class v8::Object’ has no member named ‘GetHiddenValue’; did you mean ‘IntegerValue’?
v8::Localv8::Value moduleinfo = global_obj->GetHiddenValue(SWIGV8_STRING_NEW(“swig_module_info_data”));
^~~~~~~~~~~~~~
IntegerValue
src/javascript/CMakeFiles/mraajs.dir/build.make:68: recipe for target ‘src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx.o’ failed
make[2]: *** [src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:334: recipe for target ‘src/javascript/CMakeFiles/mraajs.dir/all’ failed
make[1]: *** [src/javascript/CMakeFiles/mraajs.dir/all] Error 2
Makefile:140: recipe for target ‘all’ failed
make: *** [all] Error 2

tingleby, 2019-07-19:
So the errors come from the swig aspect of the build. Do you know what Swig version and NodeJS version you are using?

bootsy52, 2019-11-27:
you can check out my repository 

https://github.com/bootsy52/mraa

this works at least on Debian Buster on Rock Pi 4, I used armbian and kernel 5.3

the build command would be 

dpkg-buildpackage -us -uc

There were problems which I have fixed 

a) the patch for makeing yui optional does not seem work (File is unchanged) so I disabled the building of docs

b) the package nodejs-dev is not available anymore and has a new name 

c) I used clang (however gcc should work as well)

d) mraa-uart was not included in mraa-tools

bootsy52, 2019-11-27:
Forgot to mention that you still have to run 

`wget https://git.yoctoproject.org/cgit.cgi/poky/plain/meta/recipes-devtools/swig/swig/0001-Add-Node-7.x-aka-V8-5.2-support.patch
cd /usr/share/swig3.0
sudo patch -p2 <~/0001-Add-Node-7.x-aka-V8-5.2-support.patch`

from [https://github.com/intel-iot-devkit/mraa/blob/master/docs/building.md](https://github.com/intel-iot-devkit/mraa/blob/master/docs/building.md)

that still applies