eclipse-mraa/mraa #973: Can not find suitable cross - compiler on Windows with Eclipse IDE using version 2.0.0

viptuyen101, 2019-07-13:
Hi,
I've tried for 2 days to figure out that I can not find any toolchain for Windows OS that is suitable with version 2.0.0. I find out this problem when I try to use ver 1.7.0 and it works.  I've been using toolchains from [http://gnutoolchains.com/raspberry/](url) but get the error depicted in attached screenshot.
[image].

Does anyone know how to get it?

TIA,

tingleby, 2019-07-19:
Hi thanks for reaching out. Whilst this insnt a bug with mraa, i might be able to help out a bit.

Are you compiling mraa yourself with the toolchain you are using?

viptuyen101, 2019-07-22:
Hi tingleby,

Yes I built the mraa API into dynamic library (.so files) on Raspberry Pi (Linux terminal ) using Cmake version 3.13.4 and also compiled a simple example with the toolchain on Eclipse.

At first glance I thought the lastest version (the 2.0.0) may not be compatible with the toolchain because the 1.7.0 version .so file which I deploy from my colleague completely works. The thing I just found out a few days ago is that there may be something wrong with dynamic library (.so file) I built myself using Cmake to create Makefile. I have downloaded the folder of 1.7.0 version from the Releases tab and tried out the same process but the same error when compiling. 
Also I saw some errors on terminal when the Cmake was running to create Makefile
[image]

If you know the reason of problem, please let me know.
TIA,

tingleby, 2019-07-22:
Yeah so, if you want to just focus on C/C++ usages, I would disable the language bindings builds (Swig). That simplifies the build process somewhat.

Add ` -DBUILDSWIG=OFF` to your CMake invocation.

viptuyen101, 2019-07-24:
Thanks for your reply.
One more thing I would like to ask. The thing I confront firstly when I start build myself the dynamic library.
I tried these following commands on Raspberrian terminal to download and install the lib:
`sudo apt-get install git build-essential swig3.0 python-dev nodejs-dev cmake libjson-c-dev
cd /tmp
git clone https://github.com/intel-iot-devkit/mraa
cd mraa
mkdir build
cd build
cmake ..
make 
sudo make install`
By that I got the 2.0.0. After that I compile an example right on the terminal and it works. But I am using Eclipse IDE on windows PC along with toolchain from the link in my starting thread, attempt to compile an example and get the same errors as depicted. So I was running around to find out the reason.
That is the very first problem I encountered. 
I am new to these so please let me know. TIA

Propanu, 2019-07-24:
Hi TIA, is the compiler version the same on the board where you compiled MRAA, and with the toolchain you are using on your Windows host? Glibc errors are usually an indication of an incompatible/incomplete toolchain or a different/old gcc version.

viptuyen101, 2019-08-12:
Hi Propanu,
In terms of compiler's version, I come up with that using the toolchain on Windows host to build shared library instead of the native compiler on Raspberry Pi, but seem like it's not compatible with built-in content of CMakeLists.txt 
[image]
I also edited the content of CMakeLists.txt to link to the compiler and linker from toolchain which has arm-linux-gnueabihf- prefix