eclipse-mraa/mraa #978: Behavior of mraa_uart_set_flowcontrol()

lbxl2345, 2019-08-01:
If the parameter `xonxoff` is set to false, the function will call `tcflow()` and stop the device from transmitting data to the system. 

This will lead the uart example in [https://iotdk.intel.com/docs/master/mraa/python/example.html](url) fail. The function `setFlowcontrol()` will eventually call `mraa_uart_set_flowcontrol()`. Since no data is transmitted after calling `tcflow()`, the program will hang at `u.flush()`.

Is this behavior correct? 

I consider it as a bug, and I think the parameter `xonxoff` of `setFlowcontrol()` should only be used to decide whether XON/XOFF is enabled(by setting corresponding `c_iflag` of devices). Maybe another new api is needed to actually control the start/stop of data transmitting.

Propanu, 2019-09-03:
hi @lbxl2345, I think this behavior is highly depend on uart controller when using soft flow control. Besides, you may need different modes during device init vs. normal operation, so I don't think it's necessarily a bug, but I'm happy to discuss it further. Maybe I'm not fully understanding what you are trying to achieve?

In the example you are using it's specified that `u.flush()` may be unnecessary, does the example work for you without? Does it work if you leave `u.flush()` but set `xonxoff` to `True`? As far as I recall, with the latest implementation `xonxoff` parameter is used in conjunction with `rtscts` to set either soft or hard flow control. Thus one of the 2 parameters needs to be `True` respectively, when they are both `False` there is no flow control.

lbxl2345, 2019-11-04:
The example works without u.flush(). I think you're right: The behavior is highly depend the on the implementation of the controller, it's not necessarily a bug.

closed 2019-11-04 by lbxl2345

jan-kiszka, 2021-04-30:
Calling `tcflow(TCIOFF)` on `xonxoff = false` and `TCIIN` on `true` looks like a bug, not a feature. It only issues a spurious stop on the line (which was found by our system test recently). Instead, it should update `termio.c_cflag` /wrt `IXON` and `IXOFF`, the flags that `mraa_uart_settings` correctly checks on read-back. I will prepare an MR.