eclipse-mraa/mraa #980: BeagleBone Green Wireless - PWM won't initialize

fzc20070415, 2019-08-22:
When I tried to initialize Pwm, it fails:
```
>>> import mraa
>>> x = mraa.Pwm(62)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/mraa.py", line 580, in __init__
    this = _mraa.new_Pwm(pin, owner, chipid)
ValueError: Error initialising PWM on pin
```
And the system journal gives the following log:
```
Aug 22 13:31:54 beaglebone libmraa[1577]: pwm: pin 62 not initialized
```
Did I miss a step?

Alijafr, 2020-03-08:
I have the same error. Did you solved the issue?

Chuckduey, 2020-03-08:
I worked around it by writing to the /sys/class/pwm/pwmchip0/export to enable the pin.  Then /sys/class/pwm/pwm-0:0/  area to set up the period, duty cycle and then enable it.  BB overlays changed the /sys/class/pwm structure in the different releases.  So I am not using MRAA for pwm at this time.

Best of luck.