eclipse-mraa/mraa #981: Invalid gpio pin specified

thedim-witted, 2019-08-26:
I am using Arrow DragonBoard 410c to handle some interrupts on the GPIO pins. However, as I try to execute the example python script it says "Invalid GPIO pin Specified". I have tried changing the gpio pin to other number like 25, 26, 30 but nothing works. Where can I get the GPIO mapping for dragonboard 410c?

Also, I was unable to install nodejs-dev in my board due to presence of libnode-dev. I presumed that it will be necessary only if I use nodejs scripts and should not matter otherwise. Can that be the reason for this issue?

ukBaz, 2019-08-27:
You don't show what examples you are running or how you have installed mraa so it is difficult for people to debug what your issue is. As some general instructions:
GPIOs are labelled "GPIO-A" through "GPIO-L" on the DragonBoard 410c

You can look up what the pin number using mraa. For example:
```
$ sudo python3
Python 3.5.3 (default, Sep 27 2018, 17:25:39)
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa
>>> mraa.getGpioLookup('GPIO-I')
31
```

An example of an interrupt might be:
```
from signal import pause
import mraa
btn2 = mraa.Gpio(31)
btn2.dir(mraa.DIR_IN)
btn2.edge(mraa.EDGE_FALLING)
btn2.isr(mraa.EDGE_FALLING, print, 'Test on btn2')
pause()
```
How I build and install mraa for python3:
```
sudo apt install git build-essential swig3.0 python3-dev cmake

git clone https://github.com/intel-iot-devkit/mraa.git
cd mraa
mkdir build
cd build/
cmake -DBUILDSWIG=ON \
      -DBUILDSWIGPYTHON=ON \
      -DBUILDSWIGNODE=OFF \
      -DBUILDSWIGJAVA=OFF \
      -DCMAKE_INSTALL_PREFIX=/usr \
      ..
make
sudo make install
```

Hope that helps.

thedim-witted, 2019-08-27:
Hi, Thank You for your reply. I had used the general instructions given in the readme for ubuntu. While building the error I was getting was regarding nodejs-dev installation. It said that it already has libnode-dev which is replacing nodejs-dev. I tried removing libnode-dev but wasn't able to.
However, even after all this I tried importing mraa in python and it did not throw an error after which I tried the example code given in examples/python/gpio_advanced.py and it said Invalid gpio pin specified.
I thought that it is related to the nodejs because that is the only thing missing in this but I wasn't sure.
I also tried your said instructions for making but when I pass those config commands in cmake I get the following error:

 `-- INFO - NODE_ROOT_DIR is /usr/include/nodejs                  
CMake Error at /usr/share/cmake-3.13/Modules/FindPackageHandleSt
andardArgs.cmake:137 (message):                                 
  Could NOT find Nodejs (missing: NODEJS_INCLUDE_DIRS) (found ve
rsion                                                           
  "10.15.2")                                                    
Call Stack (most recent call first):                            
  /usr/share/cmake-3.13/Modules/FindPackageHandleStandardArgs.cm
ake:378 (_FPHSA_FAILURE_MESSAGE)                                
  cmake/modules/FindNodejs.cmake:95 (find_package_handle_standar
d_args)                                                         
  src/javascript/CMakeLists.txt:1 (find_package)                
                                                                
                                                                
-- Configuring incomplete, errors occurred!                     `

As it is visible it is throwing error regarding nodejs which I guess is the root problem here.

thedim-witted, 2019-08-27:
And when I tried to follow your code for the list of GPIO pins, I got the following error:

> `Python 2.7.16 (default, Apr  6 2019, 01:42:57)                   
[GCC 8.3.0] on linux2                                            
Type "help", "copyright", "credits" or "license" for more informa
tion.                                                            
>>> import mraa                                                  
>>> for i in range(ord('A'), ord('L')+1):                        
...     pin = 'GPIO-{}'.format(chr(i))                           
...     num = mraa.getGpioLookup(pin)                            
...     print(pin, num)                                          
...                                                              
Traceback (most recent call last):                               
  File "<stdin>", line 3, in <module>                            
  File "/usr/lib/python2.7/dist-packages/mraa.py", line 469, in g
etGpioLookup                                                     
    return _mraa.getGpioLookup(pin_name)                         
ValueError: Gpio name GPIO-A is not valid                        `

ukBaz, 2019-08-28:
What version of the OS are you using?

It might be worth posting a question over on the 96Boards forum for DragonBoard 410c:
https://discuss.96boards.org/c/products/dragonboard410c

ukBaz, 2019-08-28:
I have no experience of building on that OS and kernel. I am still on:
```
$ uname -a
Linux linaro-developer 4.9.56-linaro-lt-qcom #1 SMP PREEMPT Mon Oct 16 12:29:10 UTC 2017 aarch64 GNU/Linux
```

thedim-witted, 2019-08-28:
Hi,

So I was able to build the mraa into my board and I tried that PinLookup code again but it threw an error saying mraa has no attribute 'getGpioLookup'

What does that mean? I mean apart from the fact that it can't find it in spite of it ben built?

I had also tried the same code on python2.7 and 3.4

thedim-witted, 2019-08-28:
Its Debian Buster 10 with kernel version as 4.14.0-qcomlt-arm64

thedim-witted, 2019-08-28:
But I am guessing with an updated kernel version the max that would have changed is the linking of the files when they get called. And given that all the supporting files are already installed and built on the machine, it should be possible to reconfigure the link file

thedim-witted, 2019-08-28:
And one thing that I couldn't understand is why is it trying to check for linkit board when it is installed on dragonboard? I tried removing that part in the code and it shifted to the next line checking for Omega2 board. It should just stop at 96boards

thedim-witted, 2019-08-28:
> I have no experience of building on that OS and kernel. I am still on:
> 
> ```
> $ uname -a
> Linux linaro-developer 4.9.56-linaro-lt-qcom #1 SMP PREEMPT Mon Oct 16 12:29:10 UTC 2017 aarch64 GNU/Linux
> ```

BTW This is also linaro OS only. Just the kernel version is upgraded

closed 2019-08-30 by thedim-witted