eclipse-mraa/mraa #984: Anyone working on BeagleBone AI Should it be a new board?

Chuckduey, 2019-09-29:
The BeagleBone AI has been out for a while. The BegaleBone AI uses a dual-core ARM Cortex-A15. Not too bad on the CPU speed. The  BeagleBone AI has the same 2X44 pin connectors like the Black and Green.  However, the pinmux is completely different.  Some pins can have the same function but now there is up to 4 I2C buses, 4 SPI buses and 10 UART TX/RX.  The GPIO(chip)_(pin) numbering system is completely different.
The two questions are.  First is anyone working on an MRAA PR to support the new AI version?  The second question is if someone was going to build one, would it be better to put it in as a new board, or patch it in with the existing beaglebone files on MRAA?

Best regards

Chuck

tingleby, 2019-10-03:
I'm not aware of anyone working on it. 
If you do submit a PR, please send it to https://github.com/eclipse/mraa
This repo will be locked soon.

silver2row, 2020-03-26:
Hello,

Does the UPM/MRAA libraries still work w/ the BBB or BBG boards? The OS has changed dramatically in the last one to four years. 

...

I could never really get the UPM/MRAA libraries to work but I would like ot know if they are supposedly still working. 

Seth

Chuckduey, 2020-03-26:
Yes the MRAA libs are working on BBB and BBG with the exception of the PWM.  I have been running with the latest Debian 9.9 release.  Here are my steps to build:
1. sudo apt-get install build-essential python-dev cmake automake libpcre3 libpcre3-dev byacc flex 
2. git clone https://github.com/eclipse/mraa.git
3. cd mraa;mkdir build;cd build
4. cmake -D CMAKE_INSTALL_PREFIX=/usr ..
5. make
6. sudo make install
One other thing I have found is BBG takes over control of some pins with audio enabled so SPI1 can't be used.  To enable it so you can use either SPI0 bus and/or SPI1 bus disable audio in the uEnv.txt in /boot:

###Disable auto loading of virtual capes (emmc/video/wireless/adc)
#disable_uboot_overlay_emmc=1
#disable_uboot_overlay_video=1
disable_uboot_overlay_audio=1
#disable_uboot_overlay_wireless=1
#disable_uboot_overlay_adc=1
###

One other annoying thing is BB tends to use the uEnv.txt from the eMMC on boot, even with the SD card inserted. Then later it get read from the SD card for other configurations.  If you are running on the SD card, make the eMMC uEnv.txt the same as the SD card.

BB AI is a whole other board.  Some things are the same, but the pin mapping is completely different.  I have not gotten time to work on the AI version yet.

silver2row, 2020-03-27:
@Chuckduey ,

Hello Sir...this is good news. I will attempt this soon. Thank you for this update.

Seth

silver2row, 2020-04-08:
Hello @Chuckduey ,

Sir, I have found that w/ the newer image, there are some errors while compiling. I will try to test out each piece of source before I return service so there is some documented items from the 4/6/2020 image from https://elinux.org/Beagleboard:Latest-images-testing.

Seth

Chuckduey, 2020-04-08:
A few months back I did have problems getting Buster to work (Debian 10).  Currently I am sticking with Stretch version 9.9.

silver2row, 2020-04-14:
Hello...okay. I had some success but I am having issues w/ the UPM library. I will keep plugging at it.

silver2row, 2020-04-26:
Hello @Chuckduey ,

I figured out what is making the build unsuccessful so far on Buster. 

    -DBUILDSWIGNODE=OFF

At least on my machine, this has proven valuable. I have saw that there was not candidate for nodejs-dev in Buster. I am sure I can find out what took its place.

Seth

P.S. Thank you for posting.

tingleby, 2020-04-26:
Yeah, I am tempted to submit a patch setting SWIGNODE=off by default. So many issues with it :/

silver2row, 2020-04-27:
Hello @tingleby ,

Hello...do you think this could be the reasoning on why `sudo make install` is giving me errors when building? I am asking b/c make works and then errors galore happen w/ `sudo make install` on Debian 10/Buster.

...

Is there some sort of other command outside of just plain `sudo make install` that would stop the config. of NODE on my machine?

Seth

tingleby, 2020-04-30:
What error do you have after `make install`

silver2row, 2020-05-02:
Hello...I will test again soon. I will let you know. Sorry for the long delay.

silver2row, 2020-05-02:
@tingleby ,

Hello...well. This is odd. I used `sudo make install` and received no errors when not using `-DBUILDSWIGNODE=OFF` during the cmake command.

Seth

P.S. I am not sure what changed except for me not using the JSON-C library b/c I did not install it.