eclipse-mraa/mraa #986: Is anyone working on a port for stm32mp1 boards?

mgiaco, 2019-10-09:


Mani-Sadhasivam, 2019-10-09:
@mgiaco Support for Avenger96 is on my plate. If you want to contribute, please feel free to post the PR here: https://github.com/eclipse/mraa. This repository has been migrated to eclipse org.

mgiaco, 2019-10-09:
okay that sounds great... by the why is there a new repo?

mgiaco, 2019-10-09:
will there also be any support for load m4 code via python?

Mani-Sadhasivam, 2019-10-09:
> okay that sounds great... by the why is there a new repo?

Because MRAA project is now part of Eclipse IoT.

Mani-Sadhasivam, 2019-10-09:
> will there also be any support for load m4 code via python?

That is out of the scope of MRAA.

mgiaco, 2019-11-25:
@Mani-Sadhasivam Do have any timeline for that?
many thanks

Mani-Sadhasivam, 2019-11-25:
@mgiaco Sorry for the delay. Unfortunately, my time is limited these days. Will try to do it soon.

mgiaco, 2019-11-26:
Many thanks