eclipse-mraa/mraa #988: Respeaker2 core board node

wirsing, 2019-11-04:
I'm trying to get the node library running on a ReSpeaker Core v2.0 board (https://www.seeedstudio.com/ReSpeaker-Core-v2-0.html) but no luck. The developers are using the python library to talk to GPIO and it works but not on node.

Here is my code:

```
const mraa = require('mraa'); //require mraa
let power = new mraa.Gpio(12); //setup digital read on pin 12
power.dir(mraa.DIR_OUT); //set the gpio direction to output
power.write(1);
```

And here is the output:

```
let power = new mraa.Gpio(12); //setup digital read on pin 12
            ^
Error: Illegal arguments for construction of _exports_Gpio
    at Error (native)
    at Object.<anonymous> (/home/respeaker/led_custom/index.js:9:13)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.runMain (module.js:604:10)
    at run (bootstrap_node.js:390:7)
    at startup (bootstrap_node.js:150:9)
Using node 6.10.3
```

Any idea why the node version would throw this error? I tried running as root as well with no success, pin 12 should be a GPIO pin.

Propanu, 2019-11-04:
Hi @wirsing, can you please post the output of your systemd logs when initializing the pin? Usually you can do so with `journalctl`. I don't think anyone has added a board file/support for the ReSpeaker boards to the MRAA project yet so it's likely initializing as an 'unknown platform'. Technically you'd still be able to initialize GPIO pins in raw mode, but that requires the linux pin # and the API call is a bit different.

wirsing, 2019-11-04:
Hi @Propanu thanks for getting back so quickly. Here it is when starting from node, so apparently pin count is 0?:

```
Nov 04 18:10:48 v2 libmraa[2603]: libmraa version v1.9.0 initialised by user 'respeaker' with EUID 1000
Nov 04 18:10:48 v2 libmraa[2603]: Unknown Platform, currently not supported by MRAA
Nov 04 18:10:48 v2 libmraa[2603]: libmraa initialised for platform 'Unknown platform' of type 98
Nov 04 18:10:48 v2 libmraa[2603]: gpio: init: pin 12 beyond platform pin count (0)
```

Here it is when i run in python: 

```
Nov 04 18:16:46 v2 libmraa[2872]: libmraa version v1.9.0 initialised by user 'respeaker' with EUID 1000
Nov 04 18:16:46 v2 libmraa[2872]: imraa: Failed to open lock file
Nov 04 18:16:46 v2 libmraa[2872]: gpio: platform doesn't support chardev, falling back to sysfs
Nov 04 18:16:46 v2 libmraa[2872]: libmraa initialised for platform 'ReSpeaker Core v2.0' of type 19
```

Here the output of  sudo apt list --installed | grep mraa:

```
libmraa1/stretch,now 1.9.0-git20180602-0re2~re2+20180602+1 armhf [installed]
mraa-tools/stretch,now 1.9.0-git20180602-0re2~re2+20180602+1 armhf [installed]
python-mraa/stretch,now 1.9.0-git20180602-0re2~re2+20180602+1 armhf [installed]
```

Suspecting they wrote a forked version?

Propanu, 2019-11-04:
Yes, you're right, I found it here: https://github.com/respeaker/mraa
So if you use those sources to build, your node bindings should work just like the python example. I assume you built the `intel-iot-devkit` version instead?

Also, it seems like you may not need to even do a build since the `node-mraa` package is also available among the [ReSpeaker deb packages](https://github.com/respeaker/deb/blob/master/archive/dists/stretch/main/binary-armhf/Packages). I'd just try that with a `sudo apt install node-mraa`.

wirsing, 2019-11-04:
Thanks! No, I just used the npm package. Not super sure how to require when installing with apt but i guess it's just a question of finding the correct path.

Propanu, 2019-11-04:
That's right, you can simply set `NODE_PATH` to the location where `node-mraa` gets installed and that should do it. It may also help to remove the npm version. It's equivalent to the upstream version of the MRAA project that does not include support for this board.