eclipse-mraa/mraa #997: Jetson Nano Support

brsbilgin, 2020-01-24:
I have tried with jetson nano board for I2C and it has failed. Does anyone know when will come jetson nano support. Is there any way to manage it by adding source code.

Propanu, 2020-01-24:
Hi, have you tried using the "raw" I2C APIs? They allow you to map directly to an i2c bus number as exposed by the os. Take a look at the description [here](http://iotdk.intel.com/docs/master/mraa/classmraa_1_1_i2c.html#ab3124d0211ddd56435494fc1f98aa066) and also on the native C [mraa_i2c_init_raw()](http://iotdk.intel.com/docs/master/mraa/i2c_8h.html#aca34ce1fcf314b761f3bca8c121f27b4) function as it explains how the mapping works a bit better.

brsbilgin, 2020-01-25:
Dear Propanu,
my senseor is  connected on bus 1 and the output "i2cdetect -r -y 1 " is on bellow. 
agv@nvidiajetson:~/Desktop$ i2cdetect -r -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- 69 -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 

when I run the python code on bellow th output is under python code

`#! /usr/bin/python

import mraa
x=mraa.I2c(1, False )
x.address(0x69)`

Traceback (most recent call last):
  File "deneme.py", line 4, in <module>
    x=mraa.I2c(3, False )
  File "/usr/lib/python2.7/dist-packages/mraa.py", line 1245, in __init__
    this = _mraa.new_I2c(bus, raw)
ValueError: Invalid i2c bus

Propanu, 2020-01-28:
Hi @brsbilgin,
Use `x=mraa.I2c(1, True)` instead for raw bus access and you should at least be able to talk to your I2C device. To get support for the rest of the jetson nano I/Os though, someone would have to add the board definition to the project.

BobMak, 2020-08-05:
Hello @Propanu,
I have the same issue. `x=mraa.I2c(1, True)` didn't work for me.

Propanu, 2020-08-06:
Hi @BobMak, maybe the I2C bus is not enabled in your case and you have to configure your board. Does `i2cdetect` work for you?

BobMak, 2020-08-06:
@Propanu, i2cdetect works, and I can read the raw data from my sensor using the smbus2 module.

Propanu, 2020-08-11:
@BobMak, the MRAA project is also using smbus commands to handle i2c devices but the way the bus is initialized may differ from other projects that work for you. Is there any way you can provide a system log for what you tried? This can also hint at a finicky i2c chip (e.g. one that supports only a subset of i2c functions) or unstable bus.

BobMak, 2020-08-11:
@Propanu this is what I see in syslog:
```
Aug 10 19:45:59 vlad libmraa[4332]: libmraa version v2.0.0 initialised by user 'vlad' with EUID 1000          
Aug 10 19:45:59 vlad libmraa[4332]: Unknown Platform, currently not supported by MRAA                         
Aug 10 19:45:59 vlad libmraa[4332]: imraa: Failed to open lock file                                           
Aug 10 19:45:59 vlad libmraa[4332]: gpio: platform doesn't support chardev, falling back to sysfs             
Aug 10 19:45:59 vlad libmraa[4332]: libmraa initialised for platform 'Unknown platform' of type 98            
Aug 10 19:45:59 vlad libmraa[4332]: i2c_init: Selected bus 0                                                  
Aug 10 19:45:59 vlad libmraa[4332]: i2c_init: No i2c buses defined in platform
```

Propanu, 2020-08-12:
You are building this from source right? Try with `MRAAPLATFORMFORCE` set to one of the non-Intel x86 platforms when you run cmake and it should enable some I2C buses to where you can at least test the raw functions. I wouldn't use this hack for anything else than testing though, if it works I highly encourage the community to submit an actual board file.