eclipse-mraa/mraa #998: mraa does not work any more with Kernel 5.X legacy GPIO REMOVED from kernel

g-vidal, 2020-02-20:


Propanu, 2020-04-04:
@g-vidal thanks for being a long time supporter of the project. A new milestone that I see relevant and needs to be promoted more is the need to implement the io_uring APIs that were made available with more recent kernels. This will keep the project relevant and modern so I hope it addresses at least part of your concerns.

Mani-Sadhasivam, 2020-04-04:
@g-vidal We have added support for gpio chardev long ago in MRAA. That'll work with recent kernels. Please check 96boards platform support code where you'll get some reference. For switching over, you just need to provide a flag and an array of gpio chardev mapping. Finally MRAA will try to use chardev or legacy gpio depending on the kernel revision.

Let me know if you need any help.

g-vidal, 2020-04-04:
Many thanks @Propanu  and @Mani-Sadhasivam I am very busy to support teachers during the covid-19 containment with distance learning. I am deeply overbooked at the moment but I'll check ASAP and provide feedback.