eclipse-mraa/mraa #1000: mraa 2.0.0 pwm period invalid argument

novski, 2020-03-19:
I built my own openwrt and ajusted mraa to work with pwm. I can dimm a LED. But the command `logread` says `local1.err libmraa[2056]: pwm0 write_period: Failed to write to period: Invalid argument`
I tried all [period](http://iotdk.intel.com/docs/master/mraa/classmraa_1_1_pwm.html) commands with all kind of types (float and int of all sizes) but it seems not to be happy with what i feed.
Can I somehow dig deeper inside mraa? Are there other log's or debugging tools to use?

Propanu, 2020-03-24:
Hi @novski, the `period()` functions use `mraa_pwm_write_period(mraa_pwm_context dev, int period)` internally. This is part of the [pwm.c](https://github.com/eclipse/mraa/blob/master/src/pwm/pwm.c#L36) file if you care to explore the sources further. It looks like you may be dealing with a permission error. Building and debugging the project is pretty straightforward from your IDE of choice otherwise. You can generally load the project's sources and step through the code too.

novski, 2020-03-24:
Thanks @Propanu. I found an unnecessary colon in my python code. Copy left over of an javaskript example. My Bad.