eclipse-mraa/mraa #1004: API lack in chardev version

xlla, 2020-04-02:
I am try to migrate a chardev version edison mraa lib.
In my prototype, it does not work well.

I was set chardev_capable to 1 in constructor.

1. my customized adv_func->gpio_dir_post was never called, I need manipulate tristate line in this function.
2. in mraa_gpio_init_internal, no routine for chardev_capable == 1 , so my mux/bias line was not got init.  
3. how to set meaningful consumer name instead of "O".
<pre>
	line  12:      unnamed          "O"  output  active-high [used]
	line   2: "DIG2_PU_PD"       unused   input  active-high 
	line   3:  "MUX27_DIR"          "O"  output  active-high [used]
	line  14: "TRI_STATE_ALL" "O" output active-high [used]
</pre> 

my temporary hack is.
1. add call adv_func->gpio_dir_post in mraa_gpio_chardev_init.
2.  init my chardev line inside adv_func->gpio_init_internal_replace, then memcy to first param *dev,
it works but will lead memory leak, since the new created mraa_gpio_context is not release. if I free it after memcy, the program runs to segment error.
3. hardcode a name into struct gpioline_info.  I hope it can be set to program name dynamic, then I can know which app occupation that gpio line.

Propanu, 2020-04-03:
Hi @xlla, it's been a while since I last looked at the GPIO APIs but from what you are suggesting `adv_func->gpio_dir_post()` needs to be called in your case and it isn't. However, this is a function hook for setting the pin direction and it's not typically tied to init/constructor, which can be custom too by platform and features. So the best practice to use it from an application is to call the 
```
mraa_result_t mraa_gpio_dir (mraa_gpio_context dev, mraa_gpio_dir_t dir)
```
function instead with the custom defined behavior you need for the tristate. The platform chardev capability is independent of the GPIO mode you are setting for a pin.

xlla, 2020-04-04:
@Propanu , OK, I am understand. how about question 2 & 3.

xlla, 2020-04-04:
and missing hook for gpio_close_pre in chardev version.
when call mraa_gpio_close , it is just invoke _mraa_free_gpio_groups, so there is no chance to release additional resource corresponding to that pin such as bias/mux.