eclipse-mraa/mraa #1005: line flag check result wrong in chardev version

xlla, 2020-04-03:
I have init a line, and set direction to output successful.
later, when I check this line's direction by call function mraa_gpio_read_dir, it will think "Line used by kernel." and return ERROR.

in function mraa_gpio_read_dir, it will check line->flags with those macro
<pre>
#define GPIOLINE_FLAG_KERNEL		(1UL << 0) /* Line used by the kernel */
#define GPIOLINE_FLAG_IS_OUT		(1UL << 1)
#define GPIOLINE_FLAG_ACTIVE_LOW	(1UL << 2)
#define GPIOLINE_FLAG_OPEN_DRAIN	(1UL << 3)
#define GPIOLINE_FLAG_OPEN_SOURCE	(1UL << 4)
</pre>

but in mraa_gpio_chardev_dir function, it set line->flag with those macro
<pre>
/* Linerequest flags */
#define GPIOHANDLE_REQUEST_INPUT	(1UL << 0)
#define GPIOHANDLE_REQUEST_OUTPUT	(1UL << 1)
#define GPIOHANDLE_REQUEST_ACTIVE_LOW	(1UL << 2)
#define GPIOHANDLE_REQUEST_OPEN_DRAIN	(1UL << 3)
#define GPIOHANDLE_REQUEST_OPEN_SOURCE	(1UL << 4)
</pre>

if I skip mraa_gpio_read_dir and check direction,  just call mraa_gpio_write in this line, it will success, and program runs fine.

is this function right?

Propanu, 2020-04-12:
> is this function right? it will stop init in function mraa_intel_edison_fab_c statement
Hi @xlla, yes the function is right but only needed with the Arduino expansion board which is now a retired product. That pin needs to be set to MRAA_GPIO_OUT to successfully init the SPI bus and ADC.

xlla, 2020-04-13:
Hi @Propanu , I am understand it's purpose,  
the issus is in chardev version,  
  I have init a line by 
<pre>
tristate = mraa_gpio_init_by_name(TRI_STATE_ALL)
</pre>
 and set the direction by 
<pre>
mraa_gpio_dir(tristate, MRAA_GPIO_OUT)
</pre>
then I check the direction by
<pre>
 mraa_gpio_read_dir(tristate, &tristate_dir)
</pre>
it will wrong and report "Line used by kernel".