eclipse-mraa/mraa #1007: double free detected in tcache 2

xlla, 2020-04-04:
I was test ISR with mraa.
when I want quit, call mraa_gpio_close, it free dev->events, but not point to NULL,
then call _mraa_free_gpio_groups, here it will think dev->events should free again.

it seems many issue involve with gpio_chardev, is it under some test before release.

Propanu, 2020-04-12:
Hi @xlla,

>it seems many issue involve with gpio_chardev, is it under some test before release.

`gpio_chardev` support was only added for newer platforms and this happened after some earlier boards were officially retired. It was tested on those new platforms only and does not have all the advanced function hooks you saw with sysfs because those are not needed on the newer platforms. On these, GPIO muxing is done by a device driver that ships separately of libmraa.

Propanu, 2020-04-12:
Regarding `dev->events`, it's most definitely not a double free if that's what you are suggesting.
>then call `_mraa_free_gpio_groups`
does not happen, it's using `_mraa_close_gpio_event_handles` to free the events individually. Let me know if this is not what you meant though.

xlla, 2020-04-12:
@Propanu sorry for my complains, I am use an ancient platform with special mux/bias and flashing recently kernel( >= v5.5), so I have run into this odd flow.

I will try to cover that by myself.

for dev->events, I mean mraa_gpio_close will free once, inside mraa_gpio_isr_exit, will free again, then crash.

alext-mkrs, 2026-01-18:
This seems to be a peculiar corner case and an old issue at that. Closing it as I don't see what could/needs to be done here, but if you still need help, feel free to post a comment.

closed 2026-01-18 by alext-mkrs