eclipse-mraa/mraa #1009: Compilation all broken

jpelletier, 2020-04-12:
cmake .. works.
make fails.
Environment Dragonboard 410c, linaro
uname -a
[make.log.tar.gz](https://github.com/eclipse/mraa/files/4466557/make.log.tar.gz)

Linux linaro-developer 5.4.0-qcomlt-arm64 #90 SMP PREEMPT Wed Feb 12 19:09:17 UTC 2020 aarch64 GNU/Linux

Mani-Sadhasivam, 2020-04-12:
This is the SWIG compatibility issue I believe. You can disable Javascript binding generation by passing `-DBUILDSWIGNODE=OFF` to make. Else install the newer version of SWIG (>3.2.0).

This is mentioned in documentation as well: https://github.com/eclipse/mraa/blob/master/docs/building.md

jpelletier, 2020-04-12:
Thanks for the fast reply. I'll check that. I have swig 3.0.2 installed. I'll try to install 4.0.1.

jpelletier, 2020-04-12:
I installed successfully swig 4.0.1 compiled from source.
Still there's a lot of small problems and compilation gave pretty much the same errors.
With cmake .. -DBUILDSWIGNODE=OFF
the compilation and install succeeded. Thanks!

Mani-Sadhasivam, 2020-04-12:
SWIG is being a problematic area for MRAA for a long time. I hope we'll find someone who is enthusiastic enough to fix this ;)

closed 2020-04-12 by Mani-Sadhasivam