eclipse-mraa/mraa #1019: compilation broken on Dragonboard 410c

jpelletier, 2020-05-18:
Here is the result of cmake .. (as executed in build)
-- INFO - libmraa Version v2.1.0-3-ge52c5bb
-- INFO - cmake Version 3.13.4
-- INFO - Target arch is aarch64
-- INFO - Adding onewire backend support
-- INFO - Adding support for platform ALL
-- INFO - Adding support for all platforms
INFO - Adding ARM platforms
-- PYTHON2 attempting to build!
CMake Warning (dev) at /usr/share/cmake-3.13/Modules/UseSWIG.cmake:564 (message):
  Policy CMP0078 is not set.  Run "cmake --help-policy CMP0078" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  src/python/python2/CMakeLists.txt:9 (swig_add_library)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- PYTHON3 attempting to build!
CMake Warning (dev) at /usr/share/cmake-3.13/Modules/UseSWIG.cmake:564 (message):
  Policy CMP0078 is not set.  Run "cmake --help-policy CMP0078" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  src/python/python3/CMakeLists.txt:9 (swig_add_library)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- INFO - NODE_ROOT_DIR is /usr/include/nodejs
-- INFO - Padded V8 version to match SWIG format
-- INFO - Padded V8 version to match SWIG format
-- INFO - Padded V8 version to match SWIG format
-- INFO - Padded V8 version to match SWIG format
-- INFO - Padded V8 version to match SWIG format
CMake Warning (dev) at /usr/share/cmake-3.13/Modules/UseSWIG.cmake:564 (message):
  Policy CMP0078 is not set.  Run "cmake --help-policy CMP0078" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  src/javascript/CMakeLists.txt:27 (swig_add_library)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- INFO - swig Version 3.0.12
-- INFO - CXX compiler Version GNU 8.3.0
CMake Error at src/javascript/CMakeLists.txt:41 (if):
  if given arguments:

    "GREATER" "3"

  Unknown arguments specified

tingleby, 2020-05-20:
So the Swig+NodeJS stuff is not in a good shape right now, try adding `-DBUILDSWIGNODE=off` to your cmake

jpelletier, 2021-02-04:
"try adding -DBUILDSWIGNODE=off to your cmake"
This worked but another issue popped up. See #1053.

closed 2021-02-04 by jpelletier