eclipse-mraa/mraa #1023: [python] pwm resets to 0 after python script ends

peterjeschke, 2020-06-09:
I have a RockPi 4 with a simple fan connected via PWM1. When the rockpi is booted, the fan is off.

Then I run the following python script:

```python
#!/usr/bin/env python
import mraa
import time
mraa.printError(mraa.setLogLevel(7));
x = mraa.Pwm(13)
mraa.printError(x.period_us(700))
mraa.printError(x.enable(True))

mraa.printError(x.write(1))
time.sleep(5)
```

While the script is running, the fan stays off, as expected (writing 0 would enable it). However, when the script finishes, the fan suddenly turns on! I don't know why the pwm would suddenly activate. I expect the fan to stay off.

jounalctl gives nothing interesting here:

```
Jun 09 15:54:00 box libmraa[614]: libmraa version v2.0.0-35-g3f458e6 initialised by user 'root' with EUID 0
Jun 09 15:54:00 box libmraa[614]: gpio: platform doesn't support chardev, falling back to sysfs
Jun 09 15:54:00 box libmraa[614]: libmraa initialised for platform 'ROCK PI 4B' of type 20
Jun 09 15:54:00 box libmraa[614]: Loglevel 7 is set
```

The printError statements all print a success message:

```
$ sudo python pwm.py
MRAA: SUCCESS
MRAA: SUCCESS
MRAA: SUCCESS
MRAA: SUCCESS
```

Why does this happen?

peterjeschke, 2020-06-11:
Updating to 2.1 does not fix the error.

Propanu, 2020-06-11:
Hi @peterjeschke, pwm pins are set to 0 when closed, you can see this in the pwm.c file [here](https://github.com/eclipse/mraa/blob/master/src/pwm/pwm.c#L475), so this is really expected behavior. Individual MRAA platforms have the option to override this behavior though, where needed. It was implemented as such for the generic case because Buzzers and LEDs would stay on otherwise. If your fan works as 0 - ON, my suggestion would be to reinitialize the same pin as a GPIO at the end of your script and write a 1 before exit.

peterjeschke, 2020-06-15:
Thank you, @Propanu ! I understand, why this is the logical behaviour. I'll close the ticket.

closed 2020-06-15 by peterjeschke