eclipse-mraa/mraa #1027: Why isn't there a strerror for mraa_result_t?

saltyJeff, 2020-06-27:
Hi, I've been using the MRAA library with C++, so I have a lot of code that looks like:
```c++
Result result = spi.transfer(msg, resp, 4);
if(result != SUCCESS) {
	printError(result);
	throw runtime_error("MRAA Error Raised (err msg is probably above)" + to_string(result));
}
```

I was wondering why there isn't just a strerror-ish function so I can throw exceptions with the actual message inside, like
```c++
Result result = spi.transfer(msg, resp, 4);
if(result != SUCCESS) {
	throw runtime_error("MRAA Error Raised: " + mraa_strerror(result));
}
```

I looked into the definition of `printError` and there is a giant switch statement for enum to string, but it prints instead of returning the actual message. Is it deliberate not to include a strerror-type function?

Propanu, 2020-07-01:
Hi @saltyJeff, definitely not deliberate just not something that was asked up to this point. You'll notice a lot of the logging before syslog used to write to stderr. When MRAA started getting used as a service switching to syslog calls made more sense and the OS logging mechanism would then be responsible. But this is generally true for Linux builds. The `printError` function shown in the examples uses `fprintf` because not every OS build has logging enabled. This is even more so the case for MCUs and we wanted to keep the examples as portable as we can.

Would it help your current use-case if there was a dedicated function in the C++ APIs that returns an error string instead of printing it to `stdout`?

saltyJeff, 2020-07-01:
Yeah that would be really helpful for plugging it into my C++ exceptions. If you want I can do a PR for the new method (something like `mraa_strerror` or `mraa_strresult`)

Propanu, 2020-08-10:
@saltyJeff, if you have a working version and would like to get it reviewed for adding it with the next release then by all means.