eclipse-mraa/mraa #1040: A trick to compile mraa with Node 15.0.1 and Swig 4.1.0

g-vidal, 2020-10-27:
This is a followup to   #969 and remarks from @Propanu  and @nxhack , it is **not a solution but just a trick** to reach a _successfull compilation_. If anybody with the necessary competences can transform this contribution into  a true solution it will be a great step... The changes made  are validated by the compiler but  I am not sure at all that they are correct or they will yield a totally operational library.
First of all start the compilation with the last master then apply the following patch on the file `mraajsJAVASCRIPT_wrap.cxx` and restart compilation.

```unified
--- mraajsJAVASCRIPT_wrap.cxx
+++ mraajsJAVASCRIPT_wrap.cxx
@@ -782,7 +782,7 @@
 #if (V8_MAJOR_VERSION-0) < 4 && (SWIG_V8_VERSION < 0x031803)
 #define SWIGV8_STRING_NEW2(cstr, len) v8::String::New(cstr, len)
 #else
-#define SWIGV8_STRING_NEW2(cstr, len) v8::String::NewFromUtf8(v8::Isolate::GetCurrent(), cstr, v8::String::kNormalString, len)
+#define SWIGV8_STRING_NEW2(cstr, len) v8::String::NewFromUtf8(v8::Isolate::GetCurrent(), cstr, v8::NewStringType::kNormal, len).ToLocalChecked()
 #endif
 
 #if (V8_MAJOR_VERSION-0) < 4 && (SWIG_V8_VERSION < 0x031903)
@@ -823,7 +823,7 @@
 #define SWIGV8_ADJUST_MEMORY(size) v8::Isolate::GetCurrent()->AdjustAmountOfExternalAllocatedMemory(size)
 #define SWIGV8_CURRENT_CONTEXT() v8::Isolate::GetCurrent()->GetCurrentContext()
 #define SWIGV8_THROW_EXCEPTION(err) v8::Isolate::GetCurrent()->ThrowException(err)
-#define SWIGV8_STRING_NEW(str) v8::String::NewFromUtf8(v8::Isolate::GetCurrent(), str)
+#define SWIGV8_STRING_NEW(str) v8::String::NewFromUtf8(v8::Isolate::GetCurrent(), str).ToLocalChecked()
 #define SWIGV8_SYMBOL_NEW(sym) v8::String::NewFromUtf8(v8::Isolate::GetCurrent(), sym)
 #endif
 
@@ -884,7 +884,7 @@
 #define SWIGV8_TO_STRING(handle) (handle)->ToString(SWIGV8_CURRENT_CONTEXT()).ToLocalChecked()
 #define SWIGV8_NUMBER_VALUE(handle) (handle)->NumberValue(SWIGV8_CURRENT_CONTEXT()).ToChecked()
 #define SWIGV8_INTEGER_VALUE(handle) (handle)->IntegerValue(SWIGV8_CURRENT_CONTEXT()).ToChecked()
-#define SWIGV8_BOOLEAN_VALUE(handle) (handle)->BooleanValue(SWIGV8_CURRENT_CONTEXT()).ToChecked()
+#define SWIGV8_BOOLEAN_VALUE(handle) (handle)->BooleanValue(SWIGV8_CURRENT_CONTEXT()->GetIsolate())
 #define SWIGV8_WRITE_UTF8(handle, buffer, len) (handle)->WriteUtf8(v8::Isolate::GetCurrent(), buffer, len)
 #define SWIGV8_UTF8_LENGTH(handle) (handle)->Utf8Length(v8::Isolate::GetCurrent())
 #endif
@@ -1134,7 +1134,7 @@
 #elif (V8_MAJOR_VERSION-0) < 4 && (SWIG_V8_VERSION < 0x032100)
   cdata->handle.MarkIndependent(v8::Isolate::GetCurrent());
 #else
-  cdata->handle.MarkIndependent();
+  cdata->handle.SetWeak();
 #endif
 
 }
@@ -1184,8 +1184,8 @@
   }
 #endif
 
-//  v8::Handle<v8::Object> result = class_templ->InstanceTemplate()->NewInstance();
-  v8::Local<v8::Object> result = class_templ->InstanceTemplate()->NewInstance();
+   v8::Local<v8::ObjectTemplate> result_templ = class_templ->PrototypeTemplate();
+   v8::Local<v8::Object> result = result_templ->NewInstance(SWIGV8_CURRENT_CONTEXT()).ToLocalChecked();
   SWIGV8_SetPrivateData(result, ptr, info, flags);
 
   SWIGV8_ESCAPE(result);
@@ -1409,7 +1409,7 @@
 #elif (V8_MAJOR_VERSION-0) < 4 && (SWIG_V8_VERSION < 0x032100)
   cdata->handle.MarkIndependent(v8::Isolate::GetCurrent());
 #else
-  cdata->handle.MarkIndependent();
+  cdata->handle.SetWeak();
 #endif
 
   SWIGV8_ESCAPE(obj);
@@ -1441,7 +1441,7 @@
 #else  
   v8::Local<v8::Array> arr = v8::Local<v8::Array>::Cast(result);
 #endif
-  arr->Set(arr->Length(), obj);
+  arr->Set(SWIGV8_CURRENT_CONTEXT(),arr->Length(), obj);
 
   SWIGV8_ESCAPE(arr);
[... 123 lines cut ...]
   /* add static class functions and variables */
@@ -12482,14 +12482,14 @@
 
 
   /* register classes */
-  exports_obj->Set(SWIGV8_SYMBOL_NEW("uint8Array"), _exports_uint8Array_obj);
-exports_obj->Set(SWIGV8_SYMBOL_NEW("Gpio"), _exports_Gpio_obj);
-exports_obj->Set(SWIGV8_SYMBOL_NEW("I2c"), _exports_I2c_obj);
-exports_obj->Set(SWIGV8_SYMBOL_NEW("Pwm"), _exports_Pwm_obj);
-exports_obj->Set(SWIGV8_SYMBOL_NEW("Spi"), _exports_Spi_obj);
-exports_obj->Set(SWIGV8_SYMBOL_NEW("Aio"), _exports_Aio_obj);
-exports_obj->Set(SWIGV8_SYMBOL_NEW("Uart"), _exports_Uart_obj);
-exports_obj->Set(SWIGV8_SYMBOL_NEW("Led"), _exports_Led_obj);
+exports_obj->Set(SWIGV8_CURRENT_CONTEXT(), _exports_uint8Array_obj, SWIGV8_SYMBOL_NEW("uint8Array").ToLocalChecked());
+exports_obj->Set(SWIGV8_CURRENT_CONTEXT(), _exports_Gpio_obj, SWIGV8_SYMBOL_NEW("Gpio").ToLocalChecked());
+exports_obj->Set(SWIGV8_CURRENT_CONTEXT(), _exports_I2c_obj, SWIGV8_SYMBOL_NEW("I2c").ToLocalChecked());
+exports_obj->Set(SWIGV8_CURRENT_CONTEXT(), _exports_Pwm_obj, SWIGV8_SYMBOL_NEW("Pwm").ToLocalChecked());
+exports_obj->Set(SWIGV8_CURRENT_CONTEXT(), _exports_Spi_obj, SWIGV8_SYMBOL_NEW("Spi").ToLocalChecked());
+exports_obj->Set(SWIGV8_CURRENT_CONTEXT(), _exports_Aio_obj, SWIGV8_SYMBOL_NEW("Aio").ToLocalChecked());
+exports_obj->Set(SWIGV8_CURRENT_CONTEXT(), _exports_Uart_obj, SWIGV8_SYMBOL_NEW("Uart").ToLocalChecked());
+exports_obj->Set(SWIGV8_CURRENT_CONTEXT(), _exports_Led_obj, SWIGV8_SYMBOL_NEW("Led").ToLocalChecked());
 
 
   /* create and register namespace objects */
```

nxhack, 2021-04-28:
@g-vidal 

After a lot of testing, I found out that the main reason is that the handling of "SWIG_V8_VERSION" in swig is not consistent.

As a workaround, you can apply the following patch to swig.

```diff
--- a/Lib/javascript/v8/javascriptruntime.swg
+++ b/Lib/javascript/v8/javascriptruntime.swg
@@ -56,10 +56,10 @@
 %insert(runtime) %{
 #include <v8.h>
 
-#if defined(V8_MAJOR_VERSION) && defined(V8_MINOR_VERSION)
-#undef SWIG_V8_VERSION
-#define SWIG_V8_VERSION (V8_MAJOR_VERSION * 256 + V8_MINOR_VERSION)
-#endif
+//#if defined(V8_MAJOR_VERSION) && defined(V8_MINOR_VERSION)
+//#undef SWIG_V8_VERSION
+//#define SWIG_V8_VERSION (V8_MAJOR_VERSION * 256 + V8_MINOR_VERSION)
+//#endif
 
 #include <errno.h>
 #include <limits.h>
```

nxhack, 2021-04-29:
libmraa generates "SWIG_V8_VERSION" in this way.

https://github.com/eclipse/mraa/blob/master/src/javascript/binding.gyp.cmake#L14
https://github.com/eclipse/mraa/blob/master/src/javascript/binding.gyp.cmake#L37

However, there is a strange replacement in swig's javascriptruntime.swg.

https://github.com/swig/swig/blob/master/Lib/javascript/v8/javascriptruntime.swg#L59-L62

I think there is inconsistency on SWIG.

g-vidal, 2021-05-30:
Hi,
Thanks for the trick it works ... almost !
I am using to raspberry Pi3, the mraa version downloaded today (2021-05-30)  swig 4.1.0 from git master  node version v16.2.0 ; it fails with error : 
```shell
[ 38%] Building CXX object src/javascript/CMakeFiles/mraajs.dir/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx.o
In file included from /home/ens-ife/Logiciels/IoT/mraa/build/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx:1822:
/home/ens-ife/Logiciels/IoT/mraa/api/mraa/gpio.hpp: In static member function ‘static void mraa::Gpio::v8isr(uv_work_t*, int)’:
/home/ens-ife/Logiciels/IoT/mraa/api/mraa/gpio.hpp:179:63: error: no matching function for call to ‘v8::Function::Call(v8::Local<v8::Object>, int&, v8::Local<v8::Value> [1])’
  179 |         f->Call(SWIGV8_CURRENT_CONTEXT()->Global(), argc, argv);
```
If I overcome this one I get a new one :
```shell
[ 36%] Building CXX object src/javascript/CMakeFiles/mraajs.dir/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx.o
/usr/local/include/node/v8.h: In instantiation of ‘v8::Local<T>::Local(v8::Local<S>) [with S = v8::Object; T = v8::Context]’:
/home/ens-ife/Logiciels/IoT/mraa/api/mraa/gpio.hpp:182:69:   required from here
/usr/local/include/node/v8.h:210:42: error: static assertion failed: type check
  210 |     static_assert(std::is_base_of<T, S>::value, "type check");
      |                                          ^~~~~
```
I'll go to  #1064 to comment
Maybe this one could be closed ????

nxhack, 2021-05-30:
@g-vidal 

I don't have a development environment for raspberry, so this is just for looks.

Do I have the right version of gcc?
 node v16.x requires gcc and g++ version 8.3 or higher.

nxhack, 2021-05-31:
@g-vidal 

I created a development environment on a Raspberry Pi 3B with raspbian buster and tested it.

The raspbian buster is of course gcc 8.3.

There seems to be a flaw in the patch.

Until it is fixed, please build cmake with "-DCMAKE_CXX_STANDARD=14".

nxhack, 2021-05-31:
CMAKE config
```
cmake .. -DENABLEEXAMPLES=0 \
	-DBUILDSWIGNODE=ON \
	-DSWIG_EXECUTABLE=/usr/local/bin/swig \
	-DV8_ROOT_DIR=/usr/local/include/node \
	-DNODE_ROOT_DIR=/usr/local/include/node \
	-DPYTHON2LIBS_FOUND=FALSE \
	-DPYTHON2INTERP_FOUND=FALSE \
	-DFIRMATA=ON
```
cmake log
```
-- The C compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- The CXX compiler identification is GNU 8.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test HAS_C__Wall
-- Performing Test HAS_C__Wall - Success
-- Performing Test HAS_C__Werror_main
-- Performing Test HAS_C__Werror_main - Success
-- Performing Test HAS_C__Wformat
-- Performing Test HAS_C__Wformat - Success
-- Performing Test HAS_C__Wmain
-- Performing Test HAS_C__Wmain - Success
-- Performing Test HAS_C__Wuninitialized
-- Performing Test HAS_C__Wuninitialized - Success
-- Performing Test HAS_C__Winit_self
-- Performing Test HAS_C__Winit_self - Success
-- Performing Test HAS_C__Werror_implicit
-- Performing Test HAS_C__Werror_implicit - Success
-- Performing Test HAS_C__Werror_missing_parameter_type
-- Performing Test HAS_C__Werror_missing_parameter_type - Success
-- Performing Test HAS_CXX__Wall
-- Performing Test HAS_CXX__Wall - Success
-- Performing Test HAS_CXX__Werror_main
-- Performing Test HAS_CXX__Werror_main - Success
-- Performing Test HAS_CXX__Wformat
-- Performing Test HAS_CXX__Wformat - Success
-- Performing Test HAS_CXX__Wmain
-- Performing Test HAS_CXX__Wmain - Success
-- Performing Test HAS_CXX__Wuninitialized
-- Performing Test HAS_CXX__Wuninitialized - Success
-- Performing Test HAS_CXX__Winit_self
-- Performing Test HAS_CXX__Winit_self - Success
-- Performing Test HAS_CXX__Wnon_virtual_dtor
-- Performing Test HAS_CXX__Wnon_virtual_dtor - Success
-- Performing Test HAS_CXX__Woverloaded_virtual
-- Performing Test HAS_CXX__Woverloaded_virtual - Success
-- Performing Test HAS_CXX__Wreorder
-- Performing Test HAS_CXX__Wreorder - Success
[... 235 lines cut ...]
Scanning dependencies of target mraa-gpio
[ 91%] Building C object tools/CMakeFiles/mraa-gpio.dir/mraa-gpio.c.o
/home/pi/GIT/mraa/tools/mraa-gpio.c: In function ‘gpio_set’:
/home/pi/GIT/mraa/tools/mraa-gpio.c:99:13: warning: ‘mraa_gpio_use_mmaped’ is deprecated [-Wdeprecated-declarations]
             if (mraa_gpio_use_mmaped(gpio, 1) != MRAA_SUCCESS) {
             ^~
In file included from /home/pi/GIT/mraa/tools/mraa-gpio.c:13:
/home/pi/GIT/mraa/api/mraa/gpio.h:289:26: note: declared here
 DEPRECATED mraa_result_t mraa_gpio_use_mmaped(mraa_gpio_context dev, mraa_boolean_t mmap);
                          ^~~~~~~~~~~~~~~~~~~~
/home/pi/GIT/mraa/tools/mraa-gpio.c: In function ‘gpio_get’:
/home/pi/GIT/mraa/tools/mraa-gpio.c:117:13: warning: ‘mraa_gpio_use_mmaped’ is deprecated [-Wdeprecated-declarations]
             if (mraa_gpio_use_mmaped(gpio, 1) != MRAA_SUCCESS) {
             ^~
In file included from /home/pi/GIT/mraa/tools/mraa-gpio.c:13:
/home/pi/GIT/mraa/api/mraa/gpio.h:289:26: note: declared here
 DEPRECATED mraa_result_t mraa_gpio_use_mmaped(mraa_gpio_context dev, mraa_boolean_t mmap);
                          ^~~~~~~~~~~~~~~~~~~~
[ 94%] Linking C executable mraa-gpio
[ 94%] Built target mraa-gpio
Scanning dependencies of target mraa-uart
[ 97%] Building C object tools/CMakeFiles/mraa-uart.dir/mraa-uart.c.o
[100%] Linking C executable mraa-uart
[100%] Built target mraa-uart
```

g-vidal, 2021-06-01:
many thanks @nxhack 
I am using bullseye the gcc  version is OK
gcc version 10.2.1 20210110 (Debian 10.2.1-6) 

I am compiling cmake  to get cmake-3.20.3 bt it takes long I'll keep and cross check your log.

g-vidal, 2021-06-01:
@nxhack @Propanu same error occurs  when building upm

```shell
Building CXX object src/2jciebu01_ble/CMakeFiles/jsupm_2jciebu01_ble.dir/CMakeFiles/jsupm_2jciebu01_ble.dir/2jciebu01_bleJAVASCRIPT_wrap.cxx.o
In file included from /usr/local/include/node/v8.h:30,
                 from /usr/local/include/node/node.h:63,
                 from /home/ens-ife/Logiciels/IoT/upm/build/src/2jciebu01_ble/CMakeFiles/jsupm_2jciebu01_ble.dir/2jciebu01_bleJAVASCRIPT_wrap.cxx:171:
/usr/local/include/node/v8-internal.h: In function ‘void v8::internal::PerformCastCheck(T*)’:
/usr/local/include/node/v8-internal.h:452:38: error: ‘remove_cv_t’ is not a member of ‘std’; did you mean ‘remove_cv’?
  452 |             !std::is_same<Data, std::remove_cv_t<T>>::value>::Perform(data);
      |                                      ^~~~~~~~~~~
      |                                      remove_cv
/usr/local/include/node/v8-internal.h:452:38: error: ‘remove_cv_t’ is not a member of ‘std’; did you mean ‘remove_cv’?
  452 |             !std::is_same<Data, std::remove_cv_t<T>>::value>::Perform(data);
      |                                      ^~~~~~~~~~~
      |                                      remove_cv
/usr/local/include/node/v8-internal.h:452:50: error: template argument 2 is invalid
  452 |             !std::is_same<Data, std::remove_cv_t<T>>::value>::Perform(data);
      |                                                  ^
/usr/local/include/node/v8-internal.h:452:63: error: ‘::Perform’ has not been declared
  452 |             !std::is_same<Data, std::remove_cv_t<T>>::value>::Perform(data);
      |                                                               ^~~~~~~
```

nxhack, 2021-06-01:
Thanks for the report.
I will try it with gcc10. (In OpenWrt's cross build environment).

g-vidal, 2021-06-01:
OK thanks for the followup 
if it may help I keep traces of the process in my [draft  blog](https://blog.tremplin.ens-lyon.fr/GerardVidal/drafts/RaspberryPiMetBuildImage.html) without the logs and in french but code blocks are international  ;-)

nxhack, 2021-06-01:
https://blog.tremplin.ens-lyon.fr/GerardVidal/drafts/RaspberryPiMetBuildImage.html#head6.2.1
We no longer need this patch.
```diff
--- a/Lib/javascript/v8/javascriptruntime.swg
+++ b/Lib/javascript/v8/javascriptruntime.swg
@@ -57,9 +57,10 @@
 #include <v8.h>

 #if defined(V8_MAJOR_VERSION) && defined(V8_MINOR_VERSION)
-#undef SWIG_V8_VERSION
+#ifndef SWIG_V8_VERSION
 #define SWIG_V8_VERSION (V8_MAJOR_VERSION * 256 + V8_MINOR_VERSION)
 #endif
+#endif

 #include <errno.h>
 #include <limits.h>
```

g-vidal, 2021-06-01:
Could it  be the cause of the error ? 
I will restart without patching swig 

The solution is pushed in the master or should I use another branch? AT the moment I am cloning mraa master.

nxhack, 2021-06-01:
The version of gcc10 that can be configured in OpenWrt is 10.3.0, so I tested with that.
```
-- INFO - swig Version 4.1.0
-- INFO - CXX compiler Version GNU 10.3.0
-- INFO - Using V8 version > 3 so requiring C++11 compiler
```

I was able to build with gcc 10.3.0 without any problems.

That's strange...

nxhack, 2021-06-01:
> The solution is pushed in the master or should I use another branch? AT the moment I am cloning mraa master.
For swig, leave it as is 4.1.0.
Please apply my patch for mraa and upm

nxhack, 2021-06-01:
Probably not, but is swig's PTAH correct?

> [ 38%] Building CXX object src/javascript/CMakeFiles/mraajs.dir/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx.o
In file included from /home/ens-ife/Logiciels/IoT/mraa/build/src/javascript/CMakeFiles/mraajs.dir/mraajsJAVASCRIPT_wrap.cxx:1822:
/home/ens-ife/Logiciels/IoT/mraa/api/mraa/gpio.hpp: In static member function ‘static void mraa::Gpio::v8isr(uv_work_t*, int)’:
/home/ens-ife/Logiciels/IoT/mraa/api/mraa/gpio.hpp:179:63: error: no matching function for call to ‘v8::Function::Call(v8::Local<v8::Object>, int&, v8::Local<v8::Value> [1])’
  179 |         f->Call(SWIGV8_CURRENT_CONTEXT()->Global(), argc, argv);

I'm curious about this log.

Try to explicitly specify "SWIG_EXECUTABLE", which is mraa's CMAKE option.

g-vidal, 2021-06-01:
I don't think the difference between 10.2 and 10.3 might yield such discrepancy
I was wondering if this does come from the combination of 
./configure  with option --enable-cpp11-testing   and the old patch that was applied ?
The enable-cpp11-testing  sends a message in the log sayind that it tests c++17  which seems OK and it looks like just a testing option

Which patch  ? where can I find it ?
Thanks

nxhack, 2021-06-01:
I tested swig with the "--enable-cpp11-testing" option, and mraa and upm were able to build without any problems.

For more details about the patch, see

SWIG:
 This is the github master branch as it was in 4.1.0.

MRAA:
 The following patches have been applied to the master branch on github.
  https://github.com/eclipse/mraa/issues/1041 (This is the patch you created.)
  https://github.com/eclipse/mraa/pull/1064
 
UPM:
 The following patches have been applied to the master branch on github.
  https://github.com/eclipse/upm/pull/703
  https://github.com/nxhack/openwrt-custom-packages/blob/master/libupm/patches/002-at42qt1070-id.patch
  https://github.com/nxhack/openwrt-custom-packages/blob/master/libupm/patches/003-link-atomic.patch
  https://github.com/nxhack/openwrt-custom-packages/blob/master/libupm/patches/004-uint8_t.patch

g-vidal, 2021-06-02:
I have cloned swig mraa and upm from your repo (from the cleaned V8 branches)
everything run ssmoothly for all :+1: 

It seems that -DCMAKE_CXX_STANDARD=14 is crucial ! 5why not 17  insteadof 14 ?)

I'll check tomorrow to compile the eclipse master version.. maybe need to change config parameters inside CMakefiles

nxhack, 2021-06-02:
> It seems that -DCMAKE_CXX_STANDARD=14 is crucial ! 5why not 17 insteadof 14 ?)

I set the number to 14 in order to support node.js v16.
No other intentions.

g-vidal, 2021-06-02:
Things are getting clearer !!!

At least wwe know where to look :-)  your branch works master eclipse is not . Here is the situation :

tests made  using  `git clone -b master --single-branch  https://github.com/swig/swig.git` and `cmake .. -Wno-dev -DCMAKE_INSTALL_PREFIX=/usr -DENABLEEXAMPLES=0 -DBUILDSWIGNODE=ON -DSWIG_EXECUTABLE=/usr/local/bin/swig -DV8_ROOT_DIR=/usr/local/include/node -DNODE_ROOT_DIR=/usr/local/include/node 	-DPYTHON2LIBS_FOUND=FALSE 	-DPYTHON2INTERP_FOUND=FALSE`

situation 1  **FAILS** :

`git clone https://github.com/eclipse/mraa`

change in CMakefiles.txt

```shell
  cmake_policy(SET CMP0078 OLD)
  cmake_policy(SET CMP0086 NEW)
endif ()
option (BUILDSWIGNODE "Build swig node modules." ON)
option (BUILDSWIGJAVA "Build Java API." ON)
```
patch with 

```patch
--- include/version.h	2020-11-10 19:47:03.734243366 +0100
+++ include/version.h	2020-11-11 10:43:19.870180059 +0100
@@ -11,8 +11,8 @@
 extern "C" {
 #endif
 
-const char* gVERSION;
-const char* gVERSION_SHORT;
+extern const char* gVERSION;
+extern const char* gVERSION_SHORT;
 
 #ifdef __cplusplus
 }
```

Situation 2 **WORKS**  :
```git clone -b cleanup_V8_MAJOR_VERSION --single-branch  https://github.com/nxhack/mraa.git```

change in CMakefiles.txt
```shell
  cmake_policy(SET CMP0078 OLD)
  cmake_policy(SET CMP0086 NEW)
endif ()

.../...

option (BUILDSWIGNODE "Build swig node modules." ON)
option (BUILDSWIGJAVA "Build Java API." ON)
```
patch with 
```patch
--- include/version.h	2020-11-10 19:47:03.734243366 +0100
+++ include/version.h	2020-11-11 10:43:19.870180059 +0100
@@ -11,8 +11,8 @@
 extern "C" {
 #endif
 
-const char* gVERSION;
-const char* gVERSION_SHORT;
+extern const char* gVERSION;
+extern const char* gVERSION_SHORT;
 
 #ifdef __cplusplus
 }
```

g-vidal, 2021-06-02:
Hi @nxhack I checked `-DCMAKE_CXX_STANDARD=14` and `-DCMAKE_CXX_STANDARD=17` no difference ! probably we may choose 17 as it is the more uptodate?

Any idea on the orign of he difference between your branch and eclipse master ?

nxhack, 2021-06-02:
>I checked -DCMAKE_CXX_STANDARD=14 and -DCMAKE_CXX_STANDARD=17 no difference ! probably we may choose 17 as it is the more uptodate?

I think we should choose conservative options.

Again. I will explain the procedure I described in more detail.
> MRAA:
The following patches have been applied to the master branch on github.
#1041 (This is the patch you created.)
#1064

```
git clone https://github.com/eclipse/mraa.git
cd mraa
wget https://patch-diff.githubusercontent.com/raw/eclipse/mraa/pull/1064.patch
patch -p1 < 1064.patch
```
And apply your gcc10 support patch to this.

nxhack, 2021-06-03:
I recommend that you submit a pull request for the gcc10 patch.

g-vidal, 2021-06-03:
I am sorry I did not understand that one patch was not yet installed . It works like a charm !

**test OK  on Raspberry Pi3**
Linux myPi  5.12.6-v7 #1 SMP Mon May 31 18:19:07 CEST 2021 armv7l GNU/Linux
SWIG Version 4.1.0
gcc version 10.2.1 20210110 (Debian 10.2.1-6)
 node  v16.2.0
npm 7.15.0

I think that patch #1064 should be pushed in master  @Propanu

nxhack, 2021-06-03:
>I recommend that you submit a pull request for the gcc10 patch.

The same patch had already been opened.

https://github.com/eclipse/mraa/pull/1012