eclipse-mraa/mraa #1042: python mrra fails with error "NameError: name '_uint8Array__mraa' is not defined"

g-vidal, 2020-11-11:
I compile successfully mraa on rpi3 rpi3P+ rpi4 under buster  (and bullseye)
mra-XXX commands work properly but  when I try to use mraa with python3 it fails at import  :
 ```shell
raspife3P-€ : python3
Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.7/dist-packages/mraa.py", line 101, in <module>
    class uint8Array(_object):
  File "/usr/lib/python3.7/dist-packages/mraa.py", line 114, in uint8Array
    __swig_destroy__ = __mraa.delete_uint8Array
NameError: name '_uint8Array__mraa' is not defined
```
I have tried to compile 
* with g++-7  and  g++-8
* with swig 3.0.12 and 4.1.0
* mraa version 1.9.0  and 2.2
with always the same error in python  I used the same method for years now and everything was working like a charm.... Does anybody have a hint or an idea on what is wrong? 
Thanks all

g-vidal, 2020-11-12:
After many tests i tried to retrieve an old `CMakeLists.txt` and restart the compilation with it. 
MRAA works fine  no more errors  on loading 
There must be something wrong in that file. I'll try to find out but I need some sleep....

ssgnh, 2021-07-17:
how to generate the file “mraa.py”

when run my python3 file,it will return erro on "import mraa"

closed 2025-12-14 by tingleby