eclipse-mraa/mraa #1043: Using MRAA with BeagleBone green

Fabricio-Valerio, 2020-11-23:
Hello, I'm new to development using a Beaglebone board, I would like to know if MRAA can be used with BeagleBone Green or only with Beaglebone Black.

Thanks in Advance.

Chuckduey, 2020-11-23:
I have been able to use it with BB Green and Black, both wired and wireless.  
nJoy!

Fabricio-Valerio, 2020-11-23:
> I have been able to use it with BB Green and Black, both wired and wireless.
> nJoy!

Thanks for the reply @Chuckduey, I'll check the documentation and other informations to continue my project.

Fabricio-Valerio, 2020-11-23:
> > I have been able to use it with BB Green and Black, both wired and wireless.
> > nJoy!
> 
> Thanks for the reply @Chuckduey, I'll check the documentation and other informations to continue my project.
> I have to prepare my eclipse IDE, in Ubuntu , to build a project for Beaglebone Green using MRAA, I am using arm-linux-gnueabihf as GCC, but I am unable to compile with MRAA.
> 
> This is my makefile:
> [image]

I Keep getting this error: #error your machine is neither 32 bit or 64 bit ... it must be magical
[image]

Fabricio-Valerio, 2020-11-23:
[image]
I've changed the makefile now I keep getting this error. Let me explain this project: I have a project for LinkitSmart 7688, which uses MRAA, I've changed the GCC and some paths to build this project to BeagleBone Green, but now I'm getting this error.

Chuckduey, 2020-11-24:
It looks MRAA is not installed.  
Type in **mraa-gpio version** on a console to the BeagleBone.  If you have a "command not found" then you will need to install MRAA Libs.  If the version is less than 2.0 I would uninstall and download the latest from Github.  

Here are the directions for downloading and building the latest release.  Note the "apt install" usually has a very old version and may not have the latest bug fixes.
0. Load the packages do make the build **sudo apt install build-essential python-dev cmake automake libpcre3 libpcre3-dev byacc flex **
1. **git clone https://github.com/eclipse/mraa.git**
2.  Descend into the mraa directory **cd mraa**
3. Create a build work area  **mkdir build** 
4. Descend into that work area **cd build**
5. Run cmake **cmake -D CMAKE_INSTALL_PREFIX=/usr ..**
6. If cmake runs without error, run **make** 
7. If this runs without errors you are ready to install. Run **sudo make install**
8. You may need to run **sudo ldconfig** or reboot before the linker works example **gcc myfile.c -l mraa**

After this you should be able to run the command **mraa-gpio list** and see all the pins that are controllable by MRAA on your system.  

Good Luck

Chuck

Fabricio-Valerio, 2020-11-24:
> It looks MRAA is not installed.
> Type in **mraa-gpio version** on a console to the BeagleBone. If you have a "command not found" then you will need to install MRAA Libs. If the version is less than [...]

I Did the steps in my Beaglebone and in my VM (that I use to build the project).
[image]
[image]

Then, I added the /usr/lib and /usr/include to my makefile. Now I'm getting the following error 
[image]

Did I missed something? Or there's another configuration that I have to do?

Thanks in advance

Chuckduey, 2020-11-24:
You are using a virtual box on an x86 platform.  The BeagleBone is an ARM platform.  I do not know how to make Eclipse on a VM act like a BeagleBone.  You should be able to compile the program directly on the BeagleBone using GCC on that system.  For the VM and Eclipse interface you will need to find a different resource that knows more about that.

Best regards

Chuck

Fabricio-Valerio, 2020-11-25:
> You are using a virtual box on an x86 platform. The BeagleBone is an ARM platform. I do not know how to make Eclipse on a VM act like a BeagleBone. You should be able to compile the program directly [...]

Thanks Chuck. In fact, what I need is Cross-compile my code in eclipse (VM) for the BeagleBone, do you have in mind where I can get this kind of information?