eclipse-mraa/mraa #1044: Explicit object cleanup for garbage-collecting languages

jan-kiszka, 2020-11-26:
Looking for something put into the close handler of https://github.com/node-red/node-red-nodes/blob/master/hardware/intel/mraa-gpio-dout.js#L31. That shall stops that mraa nodes keep GPIOs open until the JS garbage collector finally decides to clean up. But I suppose there is nothing in the mraa API so far, is there?

And this issue should not be limited to GPIOs, correct?

How about other languages that do not support some "delete object"`, thus controlled invocation of the destructors?

tingleby, 2020-11-29:
So the language bindings are based on the CPP headers, when an object is destroyed, https://iotdk.intel.com/docs/master/mraa/gpio_8h.html#ac18f83fc915a25f06513dbd8ea0d4bad should be called. But yeah, I think maybe extra care needs to be taken in some environments.

jan-kiszka, 2020-11-30:
We are currently trying https://github.com/siemens/meta-iot2050/commit/633a77eaa49bc87d1e33c4b96f3023ca2465c115#diff-efeb60c6a49d15c96eeef5e011f89bfd8f22df93a33ecdb012407d7b5a609e7a (plus https://github.com/siemens/meta-iot2050/commit/37ca6cfd9c652c25070debb0a058df2d3ddaa3e3#diff-eec4e82a3f9112c200a581affc24bc44fa75f056ed893b3c9cda6fae2171ce28) on a customer setup, hoping that this fixes the sporadic issues seen there. If that works, I'll send an PR for that mraa patch (eventually also for the board, but that needs more work).

PR #1068 (2021-06-29): api: Add explicit close methods to classes
This is needed for bindings to languages which perform implicit and lazy
object cleanups. The explicit close methods allow to release resources
when they are no longer required, permitting deterministic reuse. One
example is node-red-node-intel-gpio which will use the new calls on node
closing.

Fixes #1044.
commit 551486ff (2021-06-29): api: Add explicit close methods to classes

This is needed for bindings to languages which perform implicit and lazy
object cleanups. The explicit close methods allow to release resources
when they are no longer required, permitting deterministic reuse. One
example is node-red-node-intel-gpio which will use the new calls on node
closing.

Fixes #1044.
closed 2021-06-29 by tingleby