eclipse-mraa/mraa #1045: Cross compile to BeagleBone Green

Fabricio-Valerio, 2020-11-27:
Hello.
I was able to build my project in the Beaglebone, after installing MRAA on it. But it isn't practical to code and build in the Beaglebone, so I tried to cross compile using a Ubuntu VM. I was able to build the application in my VM using gcc arm-linux-gnueabihf-gcc, but I was unable to build it using MRAA. This VM I used to build a project (cross compile) to Linkit Smart 7688 but I couldn't build the project to the beaglebone.

[image]

Theres is something that I'm missing?

Thanks in advance.

tingleby, 2020-11-29:
Well just looking at that, the Linker cant find libmraa, did you install it or put it in a path the Linker is looking for?

Fabricio-Valerio, 2020-11-30:
> Well just looking at that, the Linker cant find libmraa, did you install it or put it in a path the Linker is looking for?

I installed it:

[image]

And in the directory CC_TOOLS_3 is the /usr/lib and /usr/include from the beaglebone, where I installed MRAA too, I've updated the makefile and the error continues.

[image]
[image]
[image]