eclipse-mraa/mraa #1053: linking fails on Dragonboard 410c with multiple definitions

jpelletier, 2021-01-27:
version.c.o multiple definition of `gVERSION'
mraa.c.o first defined here

jpelletier, 2021-01-30:
The file version.c is generated automatically and is conflicting with the existing definition in mraa.c.
Anybody knows how to fix this?

Propanu, 2021-02-01:
Hi @jpelletier,
Have you tried disabling the tests? This might be caused by trying to build with missing dependencies:
Take a look at the [CMakeLists.txt file](https://github.com/eclipse/mraa/blob/master/CMakeLists.txt#L165).

RajithShetty619, 2021-05-29:
did u find any fix ?

jpelletier, 2021-08-28:
> did u find any fix @jpelletier?

See Propanu's comments. When disabling the tests, it works.

rei-vilo, 2022-05-20:
@jpelletier Thank you for the hint.

I solved the issue by specifying the two constants as `extern`, as the **version.h** header file only sets their declaration. The **version.c** source file provides their values.

``` c
extern const char* gVERSION;
extern const char* gVERSION_SHORT;
```

Petross404, 2022-06-13:
Can we backport this fix to the latest version?