eclipse-mraa/mraa #1057: mraa C++ bindings: printError() needs to be prefixed

mwallnoefer, 2021-02-20:
In contradiction to the example here http://iotdk.intel.com/docs/master/mraa/classmraa_1_1_led.html and the others under https://github.com/eclipse/mraa/tree/master/examples/c%2B%2B the `printError()` function needs to be prefixed with `mraa::` -> `mraa::printError()`.

Please fix the examples, thanks a lot!

mwallnoefer, 2021-02-20:
Moreover the `usleep()` function had been deprecated for quite a lot of time and should be substituted by `nanosleep()`. Please find here the respective indication in the Linux manpages:
> CONFORMING TO
>       4.3BSD,  POSIX.1-2001.   POSIX.1-2001  declares this function obsolete;
>       use nanosleep(2) instead.  POSIX.1-2008 removes  the  specification  of
>       usleep().

mwallnoefer, 2024-01-29:
Any news on this?

mwallnoefer, 2024-11-11:
Any news on this?

mwallnoefer, 2026-10-01:
Any news on this?

alext-mkrs, 2026-10-13:
I'll take a look, probably later this week.

alext-mkrs, 2026-10-27:
I've analyzed the code in question.
* On the `printError()` part - it uses the C++ standard argument-dependent lookup, so compiles and works properly. Given that it's a non-core part of the example(s), having fewer distractions in a form of the namespace prefix seems to be a very valid use case for that C++ feature to me.
* On the `usleep` vs. `nanosleep` part, indeed the former is deprecated for quite a while, but it doesn't seem like it's going away any time soon and it's simpler to use than the latter (uses a simple `int`-like parameter as opposed to the `timespec`), so again fewer distractions. Indeed though, some of the examples use `sleep()`, some use `usleep()`, and converging on one could be an option, though the benefit is purely aesthetic IMO. I'm open to reviewing a PR if anyone is willing.

mwallnoefer, 2026-10-30:
Got it. So I will propose a PR for _usleep()_ to _sleep()_ substitution where feasible.

closed 2026-10-30 by mwallnoefer