eclipse-mraa/mraa #1065: Free called twice when gpio closed with Isr enabled

nagercoil, 2021-05-19:
In mraa_gpio_close(mraa_gpio_context dev)

```
    if (dev->events) {
        free(dev->events);
    }
```

    /* Free any ISRs */
`    mraa_gpio_isr_exit(dev);`

In mraa_gpio_isr_exit(mraa_gpio_context dev)

```
if (dev->events) {
        free(dev->events);
        dev->events = NULL;
    }

```
Since the dev->events is not assigned null, double free happens.  The freeing of events should be defined inside the 
mraa_gpio_isr_exit. Two places is not correct

Propanu, 2021-05-20:
I think you're on to something, the call in `mraa_gpio_close(...)` should've been removed when c563e4fb4daebc289983ceb2f8c4e0d8f8e80e9a was merged. Initially the `gpio_close` and `gpio_isr_exit` functions were independent and managed by the user, so it did make sense to have both. Is this causing a segfault or any issues for your application?

nagercoil, 2021-05-20:
Yes it is showing error for double free. After calling free we are not assigning null. So inside the `gpio_isr_exit` when it is checked again it assumes it not freed and does the free again. 
The second issue is why we need this code in the close. The deleting of the events should be ideally inside the gpio_isr_exit.

nagercoil, 2021-05-22:
@Propanu Can I create a branch and provide the fix for the same.

Propanu, 2021-05-24:
By all means. I will also list this as a bug so it can be included with the next set of patches.

GPCExplorer, 2022-09-30:
Hi, What's the status resolving this bug?