eclipse-mraa/mraa #1066: ValueError: Invalid i2c bus when using mraa in a container

mgfink, 2021-05-20:
Hello -

I'm trying to get grove.py running on a Docker container on a Raspberry Pi 4B.  There have been several issues along the way and am at the point where I'm blocked when trying to call mraa.I2c(0).  Only on the container, it will fail with the message "libmraa[9]: i2c: Device not initialized" and other errors shown below.

I've been able to confirm general access to the GPIO pins using rpi.gpio from within the container.  I also run the container in privileged mode and pass the i2c buses as device inputs when starting the container.  I'm wondering if anyone has come across this and can point me in the right direction.

FWIW I did create a /boot/config.txt file on the container (via Dockerfile) with dtparam=i2c0=on in it.  This also did not help.

--------------- Host interaction --------------- 
pi@raspberrypi:~ $ python3
Python 3.7.3 (default, Jan 22 2021, 20:04:44)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa
>>> dev = mraa.I2c(0)
>>> print(dev)
<mraa.I2c; proxy of <Swig Object of type 'mraa::I2c *' at 0xb6703ae8> >

--------------- END --------------- 

--------------- Docker interaction --------------- 
pi@raspberrypi:~ $ docker run --device /dev/i2c-0 --device /dev/i2c-1  --privileged -it -exec grove-container bash
root@2028dd0927bd:/src# python
Python 3.7.10 (default, May 12 2021, 17:09:41)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa
libmraa[9]: libmraa version v2.2.0-3-g833bac5 initialised by user 'root' with EUID 0
libmraa[9]: gpio: platform doesn't support chardev, falling back to sysfs
libmraa[9]: libmraa initialised for platform 'Raspberry Pi Model B Rev 1' of type 5
>>> dev = mraa.I2c(0)
libmraa[9]: i2c_init: Selected bus 0
libmraa[9]: i2c: Device not initialized
libmraa[9]: i2c: If you run a kernel >=3.18 then you will have to add dtparam=i2c0=on to /boot/config.txt and reboot
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/site-packages/mraa.py", line 528, in __init__
    this = _mraa.new_I2c(bus, raw)
ValueError: Invalid i2c bus
--------------- END --------------- 

--------------- Dockerfile --------------- 

``` bash
FROM arm32v7/python:3.7-buster

ENV PYTHONUNBUFFERED 1
WORKDIR /install

# Install grove.py
RUN apt-get update && \
    apt-get install -y gcc cmake make curl gnupg git swig && \
    git clone https://github.com/Seeed-Studio/grove.py && pip install ./grove.py

# Install MRAA
RUN git clone https://github.com/eclipse/mraa.git && mkdir -p /install/mraa/build  && \
    cd /install/mraa/build && cmake -DBUILDSWIGNODE=OFF .. && \
    make && make install

# Install UPM
RUN git clone https://github.com/eclipse/upm.git && mkdir -p /install/upm/build && cd /install/upm/build && \
    cmake -DWERROR=off -DBUILDSWIGJAVA=OFF -DBUILDSWIGNODE=OFF .. && \
    make && make install

# Install remaining Python modules and cli util
RUN pip install --upgrade pip && pip install rpi.gpio
RUN wget https://project-downloads.drogon.net/wiringpi-latest.deb && dpkg -i wiringpi-latest.deb

RUN apt-get install -y i2c-tools && find /install/upm/build/ -name "*.py" -exec cp {} /usr/local/lib/python3.7/site-packages/upm/ \; && \
    mkdir /usr/local/lib/python3.7/site-packages/mraa/ && find ./mraa/build/ -name "*.py" -exec cp {} /usr/local/lib/python3.7/site-packages/mraa/ \;

WORKDIR /src
COPY src/ /src

```

--------------- END --------------- 

Thanks in advance!

mgfink, 2021-05-20:
I should probably mention that at this point I'd just like grove.py (and upm, mraa, etc.) to run in a container on a Pi4 running Raspbian Buster.  If anyone has such a thing and would be willing to share, that would be great!

Chuckduey, 2021-05-20:
I do not know about Docker, but on the Pi itself in the /boot/config.txt uses 
dtparam=i2c_vc=on 
to enable the I2C Bus 0.  Perhaps the Docker /boot/config.txt is using the Raspberry PI style notation to enable the bus.

mgfink, 2021-05-20:
That seems to have fixed it...if I added that line on the host I can run the Python sequence from there as well as from the container.  THANK YOU!!!

closed 2021-05-20 by mgfink