eclipse-mraa/mraa #1069: Nodejs for Up 2 Board: All of GPIO Pins can not be used

Jianfengusa, 2021-06-27:
Hello, 

I am using Nodejs@4.7.0 on Up 2 Board with mraa@1.0.0, I am having an issue using GPIO pins for output, can anyone help? 

Here are some error info.
[image]
[image]

Jianfengusa, 2021-06-30:
Come on guys, I really need help on this one!!!

Propanu, 2021-06-30:
Hi @Jianfengusa, upgrade to one of the latest versions of the library and it should detect your platform. UP2 was added in MRAA 1.8. Thanks!

Jianfengusa, 2021-06-30:
Thank you for the help, I will try it out.

Jianfengusa, 2021-07-01:
Thank you so much @Propanu, once I upgrade my mraa it works.

closed 2021-07-01 by Jianfengusa