eclipse-mraa/mraa #1079: HawtJNI instead of SWIG for Java bindings?

sgjava, 2021-11-28:
I'm just starting with mraa, but I've built userspace bindings with https://github.com/fusesource/hawtjni which seems superior to SWIG. For example look at my userspace library https://github.com/sgjava/java-periphery. Just wondering after looking at the limitations in https://github.com/eclipse/upm/blob/master/docs/creating_java_bindings.md. I've also built U8g2 bindings this way https://github.com/sgjava/java-u8g2.

Interested in thoughts.

g-vidal, 2021-11-29:
One of the interests of swig is to provide also nodeJS binding, for non dev users (like me)  simplest is the best even if I have to accept less performance or quality, flexibility being my first need.

sgjava, 2021-11-29:
Is there a reason you cannot use both?

tingleby, 2021-12-07:
When we first started using Swig it was mostly for the Python and the Node bindings and Java kinda come with little work for use todo. I would imagine getting both to work would be possible but not something I would be wanting to start right now. 
We would welcome any patches though :)

sgjava, 2021-12-07:
So I started out  with CFFI for Python and JNA for Java. JNI is a lot faster than JNA, but obviously with a lot more complexity. Then I found HawtJNI. If you look at https://github.com/sgjava/java-periphery you will see the Java methods make sense and there's no need for meta programming. The Armbian folks are looking for IO bindings and suggested this, but https://github.com/vsergeev/c-periphery pretty much covers all the language bases except for JS which I'm not too keen on at this point. I will however see how much work it would be to generate bindings for mraa. I will close this out for now until I have time to look at this.

closed 2021-12-07 by sgjava