eclipse-mraa/mraa #1089: Possible support for Adafruit ft232h on Linux x86 desktop

Petross404, 2022-06-13:
Hi,

I tried to use mraa (it's c++ bindings specifically) for controlling my Adafruit FT232H board and it's gpio pins but without success. Is this library supposed to work only on embedded arches or can I use it somehow on my x86 Linux for my board?

Thank you for your time.

tingleby, 2022-07-01:
This library typically runs on Linux platforms x86/arm/mips with the IO being provided by the SoC. There is support for these kind of USB break out boards, see https://github.com/eclipse/mraa/blob/master/docs/ftdi_ft4222.md which might be applicable

Petross404, 2022-10-23:
I see, I didn't know this.

Thank you very much and forgive my delayed answer.

closed 2022-10-23 by Petross404