eclipse-mraa/mraa #1099: Unable to detect GPIO pins in python

mgodbole1729, 2022-10-31:
I'm trying to access the GPIO pins via the DIO input on IEI Tank 880. I've installed the mraa library as per the instructions provided in the repo. I have also loaded the _gpio_f7188x_ kernal which is necessary for IEI  Tank. But when I try to run the code:

```
import mraa
import time

# initialise gpio 23
gpio_1 = mraa.Gpio(23)

# initialise gpio 24
gpio_2 = mraa.Gpio(24)

# set gpio 23 to output
gpio_1.dir(mraa.DIR_OUT)

# set gpio 24 to output
gpio_2.dir(mraa.DIR_OUT)

# toggle both gpio's
while True:
    gpio_1.write(1)
    gpio_2.write(0)

    time.sleep(1)

    gpio_1.write(0)
    gpio_2.write(1)

    time.sleep(1) 
```

I get the error "Invalid GPIO pin". What is going wrong here? 
I also tried different pin numbers from 0 to 8.

benbrenson, 2023-03-01:
Maybe a bit late. But what does the syslog print out? 
sudo journalctl -e -f?

frosst2906, 2024-08-08:
Hi, 

I ran into the same issue and this is what my syslog has to say. 

```
libmraa[7504]: Platform not supported, not initialising
libmraa[7504]: Checking additional Platform support for LEC-AL iPI
libmraa[7504]: gpio: platform doesn't support chardev, falling back to sysfs
libmraa[7504]: libmraa initialised for platform 'Unknown platform' of type 98
libmraa[7504]: gpio: init: pin 15 beyond platform pin count (0)
libmraa[7504]: gpio: init: pin 23 beyond platform pin count (0)
```

alext-mkrs, 2026-01-18:
Note that model 880 is not officially supported, only the 870-Q170 is, as captured in [our documentation](https://github.com/eclipse/mraa/blob/master/docs/iei-tank.md). Support would need to be added for it to work.

As this is an old issue and it's likely no longer actual, I'm closing it, but if you still need help, feel free to post a comment.

closed 2026-01-18 by alext-mkrs