eclipse-mraa/mraa #1100: IEI tank detect pins with ISMMPlus but cant detect pins with mraa

tusha-nirmal, 2022-11-01:


tusha-nirmal, 2022-11-01:
I am getting log as below
Nov  1 12:46:53 tushar-SJ28 libmraa[13174]: libmraa version v2.2.0-26-g8b1c549 initialised by user 'root' with EUID 0
Nov  1 12:46:53 tushar-SJ28 libmraa[13174]: Platform not supported, not initialising
Nov  1 12:46:53 tushar-SJ28 libmraa[13174]: Checking additional Platform support for LEC-AL iPI
Nov  1 12:46:53 tushar-SJ28 libmraa[13174]: gpio: platform doesn't support chardev, falling back to sysfs
Nov  1 12:46:53 tushar-SJ28 libmraa[13174]: libmraa initialised for platform 'Unknown platform' of type 98
Nov  1 12:47:03 tushar-SJ28 libmraa[13174]: gpio: init: pin 23 beyond platform pin count (0)

alext-mkrs, 2026-01-18:
It's unfortunately not enough data to understand what's going on, except that MRAA doesn't recognize this particular Tank board. The recognition is based on [this string](https://github.com/eclipse/mraa/blob/master/src/x86/x86.c#L97) in [this file](https://github.com/eclipse/mraa/blob/master/src/x86/x86.c#L39), so you can check if that's the case.

Note also, that "IEI Tank" is not an exact board name and there were multiple over the years, while MRAA supports specifically the 870-Q170 model, as captured in the [our documentation](https://github.com/eclipse/mraa/blob/master/docs/iei-tank.md).

As this is an old issue and it's likely no longer actual, I'm closing it, but if you still need help, feel free to post a comment.