eclipse-mraa/mraa #1121: GPIO Interrupt problem on CM3

ModnyCo-Navid, 2023-11-14:
I built the library on Debian Bullseye OS on a Radxa CM3 board, and everything goes well but when I tried to set an interrupt on any GPIO using mraa-monitor, the pin value always returns -1. I tried to run the command using sudo, but the result was the same.

tingleby, 2024-01-23:
@nascs Is this something you would be aware of?

ModnyCo-Navid, 2024-01-24:
I tried to start an ISR in c code too and I found out that before starting the ISR, function `mraa_gpio_read` returns the GPIO value correctly, but after starting an ISR by calling `mraa_gpio_isr`, the `mraa_gpio_read` will always return `-1`.
Then I stopped the ISR by calling `mraa_gpio_isr_exit` and the value returned by `mraa_gpio_read` was correct again.

nascs, 2024-01-24:
> @nascs Is this something you would be aware of?

I didn't know about this. Maybe the test missed it.

nascs, 2024-01-24:
Hi @ModnyCo-Navid , Can you show me your sample code so I can reproduce the problem and see how to fix it?

nascs, 2024-01-29:
Hi @ModnyCo-Navid, Maybe your code needs to be improved, here's an [example](https://github.com/nascs/sample_code/blob/main/mraa/c/interrupt.c) you can try, on my side it's possible to trigger to the interrupt handle function.

ModnyCo-Navid, 2024-01-29:
I can get the interrupt trigger and it works fine. There is no problem with the interrupt trigger.
My problem is that I can no longer read the pin value when I start the interrupt, and the `mraa_gpio_read` always returns `-1`.
As you can see in the sample code I sent, the read function doesn't work when I start the interrupt on pin and after exiting the interrupt, the read function works fine.

nascs, 2024-01-29:
I have reproduced this problem, and now found that there may be a problem with the parameters returned by the thread when it is created by pthread_create.